husky/husky #177: ros2 下如何将husky的urdf文件导入rviz

hxj0316, 2021-07-18:
您好，我在ros1中可以很好的将husky的urdf文件通过launch文件使得它在rviz中很好的显示，但是，在ros2中我没有尝试过，请问是否可行？如果可行具体应该如何操作？

civerachb-cpr, 2021-07-20:
I'm afraid I don't speak Chinese, so I've had to use Google Translate:

> Hello, I can pass the urdf file of Husky through the launch file in ros1 so that it can be displayed well in rviz. However, I have not tried it in ros2. Is it feasible? What should I do if it is feasible?

We are currently working on releasing ROS2 support for Husky.  It's not ready yet, but we'll make an announcement when it's ready.

> 对不起，我不会说中文，所以我用谷歌翻译把我的答案翻译回来了。 我希望这是有道理的。
> 我们目前正在努力发布对 Husky 的 ROS2 支持。 它还没有准备好，但我们会在准备好时发布公告。

tonybaltovski, 2021-11-09:
Check the `foxy-devel` branch.

closed 2022-03-16 by tonybaltovski