husky/husky #192: husky wheel odom

narutojxl, 2021-12-01:
Hello authors, 
I add `libgazebo_ros_p3d.so` into `husky.urdf.xacro` to get fake localization pose. When i use a logitech joystick to drive husky in gazebo empty.world, i find `/husky_velocity_controller/odom` topic's  `x` and `y` are very different from `/ground_truth/state` topic's `x` and `y` position.  `map` frame is the same with `odom` frame, `/husky_velocity_controller/odom` pose is in `odom` frame by default.
> 	  
      <gazebo>
	    <plugin name="p3d_base_controller" filename="libgazebo_ros_p3d.so">
	      <alwaysOn>true</alwaysOn>
	      <updateRate>20.0</updateRate>
	      <bodyName>base_link</bodyName>
	      <topicName>ground_truth/state</topicName>
	      <gaussianNoise>0.01</gaussianNoise>
	      <frameName>map</frameName>
	      <xyzOffsets>0 0 0</xyzOffsets>
	      <rpyOffsets>0 0 0</rpyOffsets>
	    </plugin>
	  </gazebo>

[image]

Then i compare  `vx` and `wz` of  `/cmd_vel`, `/husky_velocity_controller/cmd_vel`, `/husky_velocity_controller/odom`  and `/ground_truth/state`. I find `/cmd_vel`, `/husky_velocity_controller/cmd_vel` and `/husky_velocity_controller/odom` are the same in `vx` and `wz` respectively. In other word, `/husky_velocity_controller/odom` `vx` is the same with gazebo received `vx` of `/husky_velocity_controller/cmd_vel`,  `/husky_velocity_controller/odom` `wz` is the same with gazebo received `wz` of `/husky_velocity_controller/cmd_vel`.  Please see the following two pictures, the first one is  `vx` comparison chart, the second is `wz`.

My question is:
* Is  `/husky_velocity_controller/odom`  husky's wheel odom?  If it is, why the velocity `vx` and `wz` is very different from the measured  feedback velocity?
* Why husky does not track the received speed not well, as we see in the  two chart?
* Thanks for your any suggestions a lot!
[image]
[image]

destogl, 2021-12-06:
Hi @narutojxl,

You probably have in the parameters of Diff Drive controller set "open_loop" to true. This ignores the feedback from hardware and publishes odometry based on the commanded velocity.

The other issue with simulation is caused by unsuitable friction parameter on the Wheels, see #191 for the adjustments.

narutojxl, 2021-12-07:
Hello @destogl, thanks for your kind reply.  I don't find `open_loop` param in `control.yaml`. In vscode,  i also search with `open_loop`, donn't find somewhere. This is control.yaml content, is there anything i missed?  I clone from husky noetic repository.  I will merge #PR191 to my desktop noetic branch, thanks again:) 
>   
    husky_joint_publisher:
	  type: "joint_state_controller/JointStateController"
	  publish_rate: 50

    husky_velocity_controller:
	  type: "diff_drive_controller/DiffDriveController"
	  left_wheel: ['front_left_wheel', 'rear_left_wheel']
	  right_wheel: ['front_right_wheel', 'rear_right_wheel']
	  publish_rate: 50
	  pose_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.03]
	  twist_covariance_diagonal: [0.0001, 0.0001, 0.0001, 0.0001, 0.0001, 0.003]
	  cmd_vel_timeout: 0.25
	  velocity_rolling_window_size: 2

	  # Base frame_id
	  base_frame_id: base_link

	  # Odometry fused with IMU is published by robot_localization, so
	  # no need to publish a TF based on encoders alone.
	  enable_odom_tf: false

	  # Husky hardware provides wheel velocities
	  estimate_velocity_from_position: false

	  # Wheel separation and radius multipliers
	  wheel_separation_multiplier: 1.875 # default: 1.0
	  wheel_radius_multiplier    : 1.0 # default: 1.0

	  # Velocity and acceleration limits
	  # Whenever a min_* is unspecified, default to -max_*
	  linear:
	    x:
	      has_velocity_limits    : true
	      max_velocity           : 1.0   # m/s
	      has_acceleration_limits: true
	      max_acceleration       : 3.0   # m/s^2
	  angular:
	    z:
	      has_velocity_limits    : true
	      max_velocity           : 2.0   # rad/s
	      has_acceleration_limits: true
	      max_acceleration       : 6.0   # rad/s^2
*

destogl, 2022-01-18:
@narutojxl I see you are using ROS(1). Nevermind then.