husky/husky #198: How to publish GPS navigation points for outdoor navigation

miku54, 2021-12-16:
**Describe the the feature you would like**
A clear and concise description of what you want to happen.
Hi friends, I now have a set of RTK GPS with accuracy of 2cm, a multi-line radar that can be used for outdoor work, functional IMU and ODOM data, and a robot chassis. Can I use this code for outdoor navigation? How do I store and publish waypoints? I saw on YouTube that your outdoor navigation is very smooth. Is there a special software for releasing GPS navigation points?
**Other notes**
Add anything else you thing is important.

civerachb-cpr, 2021-12-16:
Hi, thanks for the question.  In its base configuration, Husky cannot be used to perform outdoor navigation.  You would need to fuse your sensor data using the robot_localization package and then add a path-generator and controller.

Fusing the sensor data is relatively simple; please refer to the robot_localization package for guidance on how to do that.  At minimum you'll likely need to add an IMU, GPS receivers, and an outdoor-rated lidar.

Regarding the path-generator and controller, that's where things get rather more difficult.  The YouTube video you mentioned uses Clearpath's closed-source, proprietary outdoor navigation software.  There are other third-party alternatives that you can experiment with, such as `move_base_flex` for the controller.  But I can't provide any good, open-source outdoor path generators, unfortunately.

miku54, 2021-12-17:
Actually, I've done the sensor fusion; But since I knew nothing about outdoor navigation, I didn't know what to do next. My friend suggested that I use Autoware, is that a good idea? Regarding the PAth-generator and controller, can MOVE_base be used for outdoor navigation? I've used it before, with amCL positioning for indoor navigation.

tonybaltovski, 2022-11-25:
Hello!

Great questions! Unfortunately this isn't quite the correct place for it, that's not something that we can inherently support. Your best bet would be to ask the ROS answers community, (https://answers.ros.org/questions/) there's usually plenty of ROS experts there that will be willing to help.

Best of luck!

closed 2022-11-25 by tonybaltovski