husky/husky #201: Focal Noetic installation instruction doesn't work on Husky

giv6kor, 2022-01-05:
- OS: Ubuntu 20.04
 - ROS Distro: Noetic
 - Installed using 'clearpath-universal-noetic-amd64-0.5.8.iso' from clearpath
 - Real hardware

 **Expected behaviour**
I followed instruction provided at [INSTALL AND CONFIGURE HUSKY SOFTWARE](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#install-and-configure-husky-software) to install and configure the Husky. Expected behavior under [Installing Husky Software](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#installing-husky-software) point 8 mentions that 'Please follow the configuration instructions on the screen. If the computer reboots, wait for the PC to boot to the login screen, and re-enter the login credentials.'  and under [Testing Base Configuration](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#testing-base-configuration) point 2 mentions that 'The COMM light on your Husky should go from red to green. You can check that the service has started correctly by checking the logs: sudo tail /var/log/upstart/ros.log -n 30'

 **Actual behaviour**
There is no configuration instruction on screen as mentioned under [Installing Husky Software](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#installing-husky-software) point 8  and unlike what is mentioned under [Testing Base Configuration](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#testing-base-configuration) point 2, COMM light changes color from red to yellow (not green) and there is no /var/log/upstart directory.
I also tried to check step 3 of [Testing Base Configuration](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#testing-base-configuration) which mentions that 'Your husky should now be accepting commands from your joystick. The service will automatically start each time you boot your Husky’s PC' whereas that also didn't work. 
 
Is [INSTALL AND CONFIGURE HUSKY SOFTWARE](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#install-and-configure-husky-software) up to date?

**To Reproduce**
Provide the steps to reproduce:
follow steps provided at [INSTALL AND CONFIGURE HUSKY SOFTWARE](https://www.clearpathrobotics.com/assets/guides/noetic/husky/InstallHuskySoftware.html#install-and-configure-husky-software)

**Other notes**
Please let me know if any other information I should add here.

civerachb-cpr, 2022-01-06:
The configuration screen mentioned was recently removed in a refactoring of the ISO; previously the hostname was assigned as `cpr-unassigned` and there was a first-boot prompt that asked for a new hostname.  This was removed and changed to a prompt during the OS installation itself.  It still applies to older versions of the ISO, but if you don't see any configuration steps after logging in that's fine.  We'll update that section of the guide.

The yellow light indicates that the Husky is either e-stopped or connected to shore power.  Please make sure the e-stop is disengaged and that the battery is properly connected.

Regarding the upstart folder, that is a copy-and-paste error from the Indigo-era manual. 18.04 and 20.04 use systemd not upstart.  The logs can be found by running

```
sudo journalctl -u ros
```

giv6kor, 2022-01-07:
Thanks for prompt response. 

Husky is not connected to power. I also checked e-stop, lock has been rotated toward unlock position and reset button has been released so in my understanding e-stop is not enabled still STOP LED is red and COMM LED is yellow. Am I missing to check something?

When I run 
`sudo journalctl -u ros`
I get following output:
```
Jan 04 13:29:35 noeticHusky systemd[1]: Started "bringup ros".
Jan 04 13:29:35 noeticHusky root[3797]: ros: Using workspace setup file /etc/ros/setup.bash
Jan 04 13:29:35 noeticHusky ros-start[3797]: <14>Jan  4 13:29:35 root: ros: Using workspace setup file /etc/ros/setup.bash
Jan 04 13:29:35 noeticHusky ros-start[3873]: ls: cannot access '/etc/ros/noetic/ros.d/*.xacro': No such file or directory
Jan 04 13:29:35 noeticHusky root[3889]: ros: Started roslaunch as background process, PID 3888, ROS_LOG_DIR=/tmp
Jan 04 13:29:35 noeticHusky ros-start[3889]: <14>Jan  4 13:29:35 root: ros: Started roslaunch as background process, PID 3888, ROS_LOG_DIR=/tmp
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-systemd-logind.service-WAeTwj': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-colord.service-WZOqFg': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-systemd-resolved.service-Zki3Mf': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-upower.service-LTKr3e': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-chrony.service-SQBRIf': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/tracker-extract-files.129': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-switcheroo-control.service-1TVByh': Permission denied
Jan 04 13:29:36 noeticHusky ros-start[3895]: /usr/bin/du: cannot read directory '/tmp/systemd-private-46aedbadacd646699441abea072f3e51-ModemManager.service-lHTkxg': Permission denied
Jan 04 13:34:15 noeticHusky systemd[1]: Stopping "bringup ros"...
Jan 04 13:34:15 noeticHusky ros-start[3940]: [WARN] [1641321255.236422]: Controller Spawner error while taking down controllers: unable to contact master
Jan 04 13:34:15 noeticHusky ros-start[3888]: WARNING: Could not change permissions for folder [/tmp/44eccf4e-6d8c-11ec-a6b3-b3e4c64188f8], make sure that the parent folder has correct permissions.
Jan 04 13:34:15 noeticHusky ros-start[3888]: Checking log directory for disk usage. This may take a while.
Jan 04 13:34:15 noeticHusky ros-start[3888]: Press Ctrl-C to interrupt
Jan 04 13:34:15 noeticHusky ros-start[3888]: Done checking log file disk usage. Usage is <1GB.
Jan 04 13:34:15 noeticHusky ros-start[3888]: 
Jan 04 13:34:15 noeticHusky ros-start[3888]: started roslaunch server http://noeticHusky:46237/
```
As per log message, it failed to access multiple directories. 
Also it says 
`cannot access '/etc/ros/noetic/ros.d/*.xacro': No such file or directory`
Can you please help me understand what could be the reason for above issues?

giv6kor, 2022-01-07:
Thanks for prompt response. 
I checked e-stop, Robot is not connected to power, lock has been rotated toward unlock position and reset button has been released so in my undestanding e-stop is not enabled still stop LED is red and COMM is yellow. Am I missing to check something?

When I run 
`sudo journalctl -u ros`
I get below output

civerachb-cpr, 2022-01-07:
Inside the battery bay there are 5 connectors: 2 red, 2 black, and 1 blue.  Can you please verify that all 5 are securely connected?

The blue connector is a short to ground, but if it is not connected it acts as an e-stop.  This is to prevent the motors from moving when the robot is connected to shore power.

giv6kor, 2022-01-10:
COMM LED is green now, it was problem with remote e-stop switch. Thanks for support.

closed 2022-01-10 by giv6kor

civerachb-cpr, 2022-01-10:
Glad you got it working!