husky/husky #214: Question about forking Husky

antoan, 2022-03-02:
Hi, 

I have adapted the Husky Kinetic codebase, for a personal project, to a low cost [mobile base](https://www.piborg.org/robots-1/monsterborg) and as I did not fork the repository directly on github, I thought I'd give you a heads up.  

At the time, my thinking was that as my hardware and setup is very different to that of Husky, there would be little to contribute back to the original repository.

My repositories based on the Husky stack can be found below: 

https://github.com/antoan/nemesis
https://github.com/antoan/nemesis_core

FYI, I have also written a high level overview of my project: here: https://antoan.github.io/Nemesis/

I have kept the original in-file licence notices that came with each package, and included your repository LICENSE file.  

Please let me know if this is OK with you or if I have missed anything out.

Cheers,

Antoan

gxfgit, 2022-03-02:
您好，邮件已经收到，我会尽快查看并回复。谢谢！

antoan, 2022-03-02:
[谢谢]

civerachb-cpr, 2022-03-02:
@antoan thanks for letting us know about your project!

All of the Husky code in this repository is released under the terms of the BSD 3-clause license, so as long as you're abiding by those terms feel free to fork the Husky code for your own needs.  I took a quick look at your Nemesis code, and everything appears to be in-order, and you're using the same license.  I don't see anything amiss.

Do you have any videos of your robot in action you can share?  I'm just curious to see it working.

antoan, 2022-03-02:
Thank you!

Sure, I've uploaded a few on a project dedicated [youtube channel](https://www.youtube.com/channel/UCMuJMMzbpIeIWIu99pVl13A)

In addition, I have some additional visual log of the project here https://www.flickr.com/photos/193485202@N04/albums/72157719550250323

Thanks again.

closed 2022-03-02 by antoan