husky/husky #215: Unable to launch husky_playpen_launch.py simulation in ROS2 Foxy

Candulf, 2022-03-12:
Hi,

I'm having trouble with running the husky_playpen_launch.py simulation.

 - OS: Ubuntu 20.04
 - ROS Distro: Foxy
 - husky packages built from source, gazebo_ros2_control installed via apt
 - Package version: 8c7d073a5a098a90011ef1c6b449d44263be8eb5 (built on March 12)
 - simulation

 **Expected behavior**
 Gazebo simulation should start, the robot can be operated via cmd_vel

 **Actual behavior**
Gazebo starts properly, the robot is spawned, but cannot be operated. The problem is connected with the gazebo_ros2_control package. I have tried to run examples provided in gazebo_ros2_control_demos to test the functionality of that package and all runs were successful. As can be seen in log messages, gazebo_ros2_control/GazeboSystem class is not found and this leads to the ros2_control_node crash. Thank you for any suggestions.

1. run: ros2 launch husky_gazebo husky_playpen_launch.py
2. error:
[INFO] [launch]: All log files can be found below /home/bestavra/.ros/log/2022-03-12-20-39-22-032413-bestavra-ThinkPad-T490-56581
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [56585]
[INFO] [ros2_control_node-2]: process started with pid [56587]
[INFO] [spawner.py-3]: process started with pid [56589]
[INFO] [gzserver-4]: process started with pid [56591]
[INFO] [gzclient-5]: process started with pid [56593]
[INFO] [spawn_entity.py-6]: process started with pid [56595]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_footprint had 0 children
[robot_state_publisher-1] Link front_bumper_link had 0 children
[robot_state_publisher-1] Link front_left_wheel had 0 children
[robot_state_publisher-1] Link front_right_wheel had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link inertial_link had 0 children
[robot_state_publisher-1] Link rear_bumper_link had 0 children
[robot_state_publisher-1] Link rear_left_wheel had 0 children
[robot_state_publisher-1] Link rear_right_wheel had 0 children
[robot_state_publisher-1] Link top_chassis_link had 0 children
[robot_state_publisher-1] Link top_plate_link had 2 children
[robot_state_publisher-1] Link top_plate_front_link had 0 children
[robot_state_publisher-1] Link top_plate_rear_link had 0 children
[robot_state_publisher-1] Link user_rail_link had 0 children
[robot_state_publisher-1] [INFO] [1647113962.317929354] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1647113962.318082749] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1647113962.318100496] [robot_state_publisher]: got segment front_bumper_link
[robot_state_publisher-1] [INFO] [1647113962.318109802] [robot_state_publisher]: got segment front_left_wheel
[robot_state_publisher-1] [INFO] [1647113962.318118723] [robot_state_publisher]: got segment front_right_wheel
[robot_state_publisher-1] [INFO] [1647113962.318127227] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-1] [INFO] [1647113962.318135357] [robot_state_publisher]: got segment inertial_link
[robot_state_publisher-1] [INFO] [1647113962.318143451] [robot_state_publisher]: got segment rear_bumper_link
[robot_state_publisher-1] [INFO] [1647113962.318151640] [robot_state_publisher]: got segment rear_left_wheel
[robot_state_publisher-1] [INFO] [1647113962.318160229] [robot_state_publisher]: got segment rear_right_wheel
[robot_state_publisher-1] [INFO] [1647113962.318168203] [robot_state_publisher]: got segment top_chassis_link
[robot_state_publisher-1] [INFO] [1647113962.318176131] [robot_state_publisher]: got segment top_plate_front_link
[robot_state_publisher-1] [INFO] [1647113962.318185338] [robot_state_publisher]: got segment top_plate_link
[robot_state_publisher-1] [INFO] [1647113962.318193838] [robot_state_publisher]: got segment top_plate_rear_link
[robot_state_publisher-1] [INFO] [1647113962.318201909] [robot_state_publisher]: got segment user_rail_link
[ros2_control_node-2] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[ros2_control_node-2]   what():  According to the loaded plugin descriptions the class gazebo_ros2_control/GazeboSystem with base class type hardware_interface::SystemInterface does not exist. Declared types are  fake_components/GenericSystem husky_base/HuskyHardware test_hardware_components/TestSystemCommandModes test_hardware_components/TestTwoJointSystem test_system
[ERROR] [ros2_control_node-2]: process has died [pid 56587, exit code -6, cmd '/opt/ros/foxy/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_gh9fcccx --params-file /home/bestavra/ros2_wss/foxy_wss/bachelor_ws/install/husky_control/share/husky_control/config/control.yaml'].
[spawner.py-3] [INFO] [1647113962.569515695] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawn_entity.py-6] [INFO] [1647113962.748297691] [spawn_husky]: Spawn Entity started
[spawn_entity.py-6] [INFO] [1647113962.748634691] [spawn_husky]: Loading entity published on topic robot_description
[spawn_entity.py-6] [INFO] [1647113962.749733552] [spawn_husky]: Waiting for entity xml on robot_description
[spawner.py-3] [INFO] [1647113964.576616902] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawn_entity.py-6] [INFO] [1647113966.403234685] [spawn_husky]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-6] [INFO] [1647113966.403998892] [spawn_husky]: Waiting for service /spawn_entity
[spawn_entity.py-6] [INFO] [1647113966.413502725] [spawn_husky]: Calling service /spawn_entity
[spawner.py-3] [INFO] [1647113966.586544201] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawn_entity.py-6] [INFO] [1647113966.589821109] [spawn_husky]: Spawn status: SpawnEntity: Successfully spawned entity [husky]
[gzserver-4] [INFO] [1647113966.618622040] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[... 26 lines cut ...]
[gzserver-4] [INFO] [1647113966.637714826] [gazebo_ros2_control]: 		 velocity
[gzserver-4] [INFO] [1647113966.637721018] [gazebo_ros2_control]: 	State:
[gzserver-4] [INFO] [1647113966.637727476] [gazebo_ros2_control]: 		 position
[gzserver-4] [INFO] [1647113966.637733562] [gazebo_ros2_control]: 		 velocity
[gzserver-4] [INFO] [1647113966.637951081] [gazebo_ros2_control]: Loading controller_manager
[INFO] [spawn_entity.py-6]: process has finished cleanly [pid 56595]
[gzserver-4] [WARN] [1647113966.648544887] [gazebo_ros2_control]:  Desired controller update period (0.1 s) is slower than the gazebo simulation period (0.01 s).
[gzserver-4] [INFO] [1647113966.648646889] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[gzserver-4] [INFO] [1647113966.791068702] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner.py-3] [INFO] [1647113966.849689702] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[gzserver-4] [INFO] [1647113966.850548545] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[spawner.py-3] [INFO] [1647113967.050831349] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[INFO] [spawner.py-3]: process has finished cleanly [pid 56589]
[INFO] [spawner.py-7]: process started with pid [56779]
[spawner.py-7] [INFO] [1647113967.464013351] [spawner_husky_velocity_controller]: Waiting for /controller_manager services
[gzserver-4] [INFO] [1647113967.674120899] [controller_manager]: Loading controller 'husky_velocity_controller'
[spawner.py-7] [INFO] [1647113967.751813271] [spawner_husky_velocity_controller]: Loaded husky_velocity_controller
[gzserver-4] [INFO] [1647113967.753058732] [controller_manager]: Configuring controller 'husky_velocity_controller'
[spawner.py-7] [INFO] [1647113967.953735895] [spawner_husky_velocity_controller]: Configured and started husky_velocity_controller
[INFO] [spawner.py-7]: process has finished cleanly [pid 56779]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1647114339.436527569] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 56585]
[INFO] [gzserver-4]: process has finished cleanly [pid 56591]
[INFO] [gzclient-5]: process has finished cleanly [pid 56593]

alpylmz, 2022-03-14:
Hi, 

I am also having trouble, but with a different error. 

I have the same environment and system, I am executing the same command, but the command fails with following error.

[INFO] [launch]: All log files can be found below /home/alp/.ros/log/2022-03-14-12-51-25-750064-alp-R2D2-27273
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("executed command failed. Command: /opt/ros/noetic/bin/xacro /home/alp/husky_ws/install/husky_description/share/husky_description/urdf/husky.urdf.xacro name:=husky prefix:='' is_sim:=true gazebo_controllers:=/home/alp/husky_ws/install/husky_control/share/husky_control/config/control.yaml")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 3 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 422, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 382, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 145, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 69, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/command.py", line 116, in perform
    raise SubstitutionFailure(f'executed command failed. Command: {command_str}')
launch.substitutions.substitution_failure.SubstitutionFailure: executed command failed. Command: /opt/ros/noetic/bin/xacro /home/alp/husky_ws/install/husky_description/share/husky_description/urdf/husky.urdf.xacro name:=husky prefix:='' is_sim:=true gazebo_controllers:=/home/alp/husky_ws/install/husky_control/share/husky_control/config/control.yaml

tonybaltovski, 2022-03-15:
@Candulf for ros2_controls you need to publish to `/husky_velocity_controller/cmd_vel_unstamped`

Candulf, 2022-03-16:
@tonybaltovski It works, thank you. My bad I haven't listed the published topics first.

alpylmz, 2022-03-16:
@Candulf It worked! Thank you.

closed 2022-03-16 by tonybaltovski

alpylmz, 2022-03-22:
> @Candulf for ros2_controls you need to publish to `/husky_velocity_controller/cmd_vel_unstamped`

I'm afraid I did not understand what you said. I changed every `/husky_velocity_controller/cmd_vel` to `/husky_velocity_controller/cmd_vel_unstamped` in the workspace, but still I cannot launch husky the same way in the first comment. What am I doing wrong?