husky/husky #218: Robot NOT turning providing angular velocity

marco-ambrosio, 2022-03-22:
**Please provide the following information:**
 - OS: **Ubuntu 20.04**
 - ROS Distro: **Foxy**
 - Built from source or installed: **Built from source**
 - Package version: **commit** 00d81bfce3222cb0c4b5c6565a92054cc2df9c46 
 - Real hardware or simulation: **Real hardware**

 **Expected behaviour**
 When an angular velocity is provided robot is expected to turn.

 **Actual behaviour**
The robot instead slowly move forward or backward in a straight line

**To Reproduce**
Provide the steps to reproduce:
1. ros2 launch husky_base base_launch.py
2. provide an angular velocity on the cmd_vel topic
3. the robot moves in a straight line

tonybaltovski, 2022-05-06:
@marco-ambrosio do you get any warnings or errors when running `ros2 launch husky_base base_launch.py`?

Ronoman, 2022-09-12:
I am experiencing the same issue. I have 20.04 running within a docker container, running ROS2 Galactic. I have built this package from source, with a few small modifications off of the galactic-devel branch (unfortunately I can't share these, but none of them should impact this).

Here is the console log from running `husky_base/base_launch.py`: https://gist.github.com/Ronoman/18c0f57437b0b491db7554c2b3c1cde1

I have echoed `/husky_velocity_controller/cmd_vel_unstamped`, and it is receiving `Twist` messages with both a linear x and angular z velocity. However, like @marco-ambrosio , the robot only drives backwards and forwards, no turning.

My fork is based off of the most recent commit on `galactic-devel` from 4 months ago, `5638ddaa`.

Ronoman, 2022-09-13:
Fixed in https://github.com/husky/husky/pull/246.

tonybaltovski, 2022-09-14:
@Ronoman can you checkout the latest `galacitc-devel`, I cherry-picked the fix for this from foxy.  As a note, Clearpath does really intend to support Galactic since it is not a LTS version.

closed 2022-10-11 by tonybaltovski