husky/husky #229: ROS2 Foxy Gazebo Simulation Crashes

robertokcanale, 2022-04-29:
I; using Ubuntu 20 with foxy. 

I just followed the tutorial provided here  [https://clearpathrobotics.com/assets/guides/foxy/husky/HuskySimulation.html](https://clearpathrobotics.com/assets/guides/foxy/husky/HuskySimulation.html ) and  RVIZ doesn't work. There is an error from the front and left wheel transform and it does not seem to work. Please refer to the image attached . 

[image]

(FYI, there is a typo on the launch example provided on the tutorial page _ros2 launch husky_viz view_robot.launch.py_ _ --->_  _ros2 launch husky_viz view_robot_launch.py_   )))

OUTPUT:

[rviz2-1] Warning: Invalid frame ID "rear_left_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "rear_right_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "front_left_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "front_right_wheel" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp

Provide the steps to reproduce:
1.  ros2 launch husky_gazebo husky_playpen_launch.py

2. ros2 launch husky_viz view_robot.launch.py

mikepurvis, 2022-04-29:
Do you have a `joint_states` topic? That's what robot_state_publisher consumes in order to publish the correct positions of all the joints, including the wheels, see:

http://wiki.ros.org/robot_state_publisher

tonybaltovski, 2022-04-29:
Also, could you provide the full log of when you start the simulation as well as the version of husky software you using?

robertokcanale, 2022-05-01:
@mikepurvis no i do not have a joint_states topic! is it already in the husky package, and if so, which one and which executable?

 or should I provide it myself? Thank you for the info!

@tonybaltovski At the moment I am trying to use the ROS2 simulation husky, not the real one. I do not know which husky version we have in the lab (it has been there for a couple of years at least), but it would be interesting to know which versions are ROS2 compatible and which are not. Where is it possible to check it?

tonybaltovski, 2022-05-03:
I mean the version of the husky software.

robertokcanale, 2022-05-04:
The husky software I am using is the one for Foxy

tonybaltovski, 2022-05-04:
Did you install it from source or via debian?  If from source, which commit?

closed 2022-10-14 by tonybaltovski