husky/husky #230: Expand ROS2 Tutorials

robertokcanale, 2022-04-29:
Hello, I will start a ROS2 project with the Husky and a UR5 mounted on top of it. First we simulate it then i'll need to make the real project.
The tutorials provided for ROS2 are very scarce. Would it be possible to have a more in-depth manual and/or more tutorials on control and simulation? 
Thank you very much,
Roberto

tonybaltovski, 2022-04-29:
Hi @robertokcanale,
We are working on expanding accessory support for the Husky but currently there is not Movieit/manipulator support.

tonybaltovski, 2024-01-02:
For ROS 2 Humble, this repository is no longer used. Please visit [clearpath_common](https://github.com/clearpathrobotics/clearpath_common). Check the [documentation](https://docs.clearpathrobotics.com/docs/ros/) for more details including supported sensors.

closed 2024-01-02 by tonybaltovski