husky/husky #236: Bring back explicit variables for laser_enabled, realsense_enabled

wakoko79, 2022-05-25:
Bring this back, this is ridiculous.
I'm on melodic and using the published husky packages (0.4.12) BREAKS THE TUTORIALS.

This commit [12230e4d7bd9356e09fbf562f5cd8493128dcb31](https://github.com/husky/husky/commit/12230e4d7bd9356e09fbf562f5cd8493128dcb31) breaks out of the box operation of laser scans in simulator, which leads to amcl not publishing map tf frame. 
And so, the tutorials are now useless. I guess this and more recent commits were merged to melodic-devel.

At least update the tutorial to indicate the variables needed to be included to make it work again.

wakoko79, 2022-05-25:
The actual commit for melodic-devel is [425e5a96d181fa9009811949ff0b5afb8b10e524](https://github.com/husky/husky/commit/425e5a96d181fa9009811949ff0b5afb8b10e524).

The commit message says: "Remove the need to explicitly specify the laser_enabled, ...".
This is wrong, as previously, there is **no need** to specify them as arguments, because you can either specify those in the launch file as args **OR** as env vars, or even don't set them at all; the tutorials will still work. But now, new people will absolutely have no idea what is wrong.

Now you have no choice but to use env vars. You can only effectively use env vars for parametrizing  when you set them outside ROS, otherwise args are better from a devel standpoint.

tonybaltovski, 2022-05-31:
Hi,
Thank you for catching this and sorry for the frustration it caused you.  The tutorials will be updated shortly.  Environment variables work better for more complex integrations involving moveit for example.

civerachb-cpr, 2022-05-31:
For context, the primary issue that led to the merge request in question was that on our physical robots we use `/etc/ros/setup.bash` to set the environment variables for adding common accessories to the Husky. But because of the previous structure of the URDF and launch files when you used these envars and tried running the MoveIt Setup Assistant, the envar-controlled accessories would not be added to the robot, resulting in missing links.

By moving to an envar-only system we fixed that issue and made it easier to run simulations that re-use the same environment-variable-exporting script as the real robot. Instead of needing to set multiple launch arguments you can simply copy the setup.bash file off of the physical robot, source it, and launch Gazebo to get the simulated robot to run with the same sensors and accessories as the real robot has.

shrinivas96, 2022-09-02:
Hello. I am facing a similar problem I think. I cloned the entire husky repository and built the `husky_gazebo`, `husky_viz` and `husky_navigation` suceesfully to try the [Husky AMCL demo tutorial](http://wiki.ros.org/husky_navigation/Tutorials/Husky%20AMCL%20Demo). But i get two errors constantly, which causes the demo to not work:

1. AMCL node complains: **No laser scan received (and thus no pose updates have been published)**
I checked, nothing is being published to the `/scan` topic (because I do not see a laser scanner mounted on top of husky in the first place)

2. move_base node complains: **Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.**

I would appreciate any help in this case. Many thanks in advance.

wakoko79, 2022-09-02:
I just checked and they still haven't updated their tutorials in Clearpath's website (I think). I think most of the people who would simulate Husky will probably be your customer, don't do this to your customers please. Please update your stuff, This is documentation, not code, it shouldn't be too hard to do.

### TLDR for @tonybaltovski:
Before launching the simulation in gazebo, your command line needs to have the environment variable, so type in your console (same console in which you will spawn husky on): `HUSKY_LMS1XX_ENABLED=1`.

**_This will not actually fix your AMCL demo if you have same husky packages as I do._**
On my end, this publishes laser scan onto `/front/scan` instead on `/scan` which AMCL subscribes to. So you have to remap which topic AMCL subsribes to for laser scans    

### Longer stuff:
The maintainers decided to ditch roslaunch args for the Husky URDF and exclusively use environment variables. The main issue is in `husky_description/launch/description.launch`.
Prior the commit I mentioned originally, `description.launch` passes args from its caller down to `husky_description/urdf/husky.urdf.xacro` which in turn enables what attachments husky has in the simulation.

Now that the args are not there, the urdf.xacro will only know that you want a particular attachment to shown up when you have its specific environment variable set in your console. So read `husky_description/urdf/husky.urdf.xacro`, there is a bunch of env vars in there that you can set.

To make it less painful for myself I just made a script like this:
```
#!/bin/bash
# Source this file to for husky gazebo env variables (starting from husky 0.4.9)
# Review husky_description/urdf/husky.urdf.xacro for related env vars.
set -a
HUSKY_LMS1XX_ENABLED=1
# HUSKY_LMS1XX_SECONDARY_ENABLED=1
# HUSKY_UST10_ENABLED=1
# HUSKY_UST10_SECONDARY_ENABLED=1
# HUSKY_LASER_3D_ENABLED=1              
# HUSKY_REALSENSE_ENABLED=1
# HUSKY_FRONT_BUMPER_EXTEND
# HUSKY_REAR_BUMPER_EXTEND
# HUSKY_URDF_EXTRAS

set +a

```

You then either source this on the console in which you spawn husky, or source it in your `~/.bashrc` so you have env vars on all console you open.  
Or you know, just straight up put `HUSKY_LMS1XX_ENABLED=1` in your `~/.bashrc` if you want it all to be automatic.  
  
But I like it sourced manually only when I need it, since it may give you headache later on when you forgot that you put it in your `~/.bashrc`.

shrinivas96, 2022-09-09:
Thanks a lot @wakoko79 for such a detailed explanation. Your explanation and some exploration of these files also helped me in better understanding the package, and how to change things that I need.

Also, yes, I agree, they should update the tutorials at this point to reflect all these changes.

hyoungmin94, 2022-10-06:
[image]
I'm having the same problem, and putting HUSKY_LMS1XX_ENABLED=1 in my ~/.bashrc didn't change anything. Any help would be appreciated.

mateusmenezes95, 2022-12-04:
Based on the suggestion made by @wakoko79 (thanks), I took all the env vars from the URDF and put them in a bash script. Here is the file content:

```sh
#!/bin/bash
# Source this file to for husky gazebo env variables (starting from husky 0.4.9
# Review husky_description/urdf/husky.urdf.xacro for related env vars.
set -a

# IMU Link
# HUSKY_IMU_XYZ=     # default: 0.19 0 0.149
# HUSKY_IMU_RPY=     # default: 0 -1.5708 3.1416
# HUSKY_IMU_PARENT=  # default: base_link

# LMS1XX Laser Primary and Secondary
# HUSKY_LMS1XX_ENABLED= # default: 0
# HUSKY_LMS1XX_TOPIC=   # default: front/scan
# HUSKY_LMS1XX_TOWER=   # default: 1
# HUSKY_LMS1XX_PREFIX=  # default: front
# HUSKY_LMS1XX_PARENT=  # default: top_plate_link
# HUSKY_LMS1XX_XYZ=     # default: 0.2206 0.0 0.00635
# HUSKY_LMS1XX_RPY=     # default: 0.0 0.0 0.0

# HUSKY_LMS1XX_SECONDARY_ENABLED=  # default: 0
# HUSKY_LMS1XX_SECONDARY_TOPIC=    # default: rear/scan
# HUSKY_LMS1XX_SECONDARY_TOWER=    # default: 1
# HUSKY_LMS1XX_SECONDARY_PREFIX=   # default: rear
# HUSKY_LMS1XX_SECONDARY_PARENT=   # default: top_plate_link
# HUSKY_LMS1XX_SECONDARY_XYZ=      # default: -0.2206 0.0 0.00635
# HUSKY_LMS1XX_SECONDARY_RPY=      # default: 0.0 0.0 3.14159

# UST10 Laser Primary  and Secondary
# HUSKY_UST10_ENABLED=  # default: 0
# HUSKY_UST10_TOPIC=    # default: front/scan
# HUSKY_UST10_PREFIX=   # default: front
# HUSKY_UST10_PARENT=   # default: top_plate_link
# HUSKY_UST10_XYZ=      # default: 0.2206 0.0 0.00635
# HUSKY_UST10_RPY=      # default: 0 0 0

# HUSKY_UST10_SECONDARY_ENABLED=  # default: 0
# HUSKY_UST10_SECONDARY_TOPIC=    # default: rear/scan
# HUSKY_UST10_SECONDARY_PREFIX=   # default: rear
# HUSKY_UST10_SECONDARY_PARENT=   # default: top_plate_link
# HUSKY_UST10_SECONDARY_XYZ=      # default: -0.2206 0.0 0.00635
# HUSKY_UST10_SECONDARY_RPY=      # default: 0 0 3.14159

# Velodyne LiDAR Primary and Secondary
# HUSKY_LASER_3D_ENABLED=  # default: 0
# HUSKY_LASER_3D_TOPIC=    # default: points
# HUSKY_LASER_3D_TOWER=    # default: 1
# HUSKY_LASER_3D_PREFIX=   # default: 
# HUSKY_LASER_3D_PARENT=   # default: top_plate_link
# HUSKY_LASER_3D_XYZ=      # default: 0 0 0
# HUSKY_LASER_3D_RPY=      # default: 0 0 0

# HUSKY_LASER_3D_SECONDARY_ENABLED=  # default: 0
# HUSKY_LASER_3D_SECONDARY_TOPIC=    # default: secondary_points
# HUSKY_LASER_3D_SECONDARY_TOWER=    # default: 1
# HUSKY_LASER_3D_SECONDARY_PREFIX=   # default: secondary_
# HUSKY_LASER_3D_SECONDARY_PARENT=   # default: top_plate_link
# HUSKY_LASER_3D_SECONDARY_XYZ=      # default: 0 0 0
# HUSKY_LASER_3D_SECONDARY_RPY=      # default: 0 0 -3.14159

# Realsense Camera Primary and Secondary
# HUSKY_REALSENSE_ENABLED=  # default: 0
# HUSKY_REALSENSE_TOPIC=    # default: realsense
# HUSKY_REALSENSE_PREFIX=   # default: camera
# HUSKY_REALSENSE_PARENT=   # default: top_plate_link
# HUSKY_REALSENSE_XYZ=      # default: 0 0 0
# HUSKY_REALSENSE_RPY=      # default: 0 0 0

# HUSKY_REALSENSE_SECONDARY_ENABLED=  # default: 0
# HUSKY_REALSENSE_SECONDARY_TOPIC=    # default: realsense_secondary
# HUSKY_REALSENSE_SECONDARY_PREFIX=   # default: camera_secondary
# HUSKY_REALSENSE_SECONDARY_PARENT=   # default: top_plate_link
# HUSKY_REALSENSE_SECONDARY_XYZ=      # default: 0 0 0
# HUSKY_REALSENSE_SECONDARY_RPY=      # default: 0 0 0

# BlackflyS Camera Primary and Secondary
# HUSKY_BLACKFLY=                #default: 0
# HUSKY_BLACKFLY_MOUNT_ENABLED=  #default: 1
# HUSKY_BLACKFLY_MOUNT_ANGLE=    #default: 0
# HUSKY_BLACKFLY_PREFIX=         #default: blackfly
# HUSKY_BLACKFLY_PARENT=         #default: top_plate_link
# HUSKY_BLACKFLY_XYZ=            #default: 0 0 0
# HUSKY_BLACKFLY_RPY=            #default: 0 0 0

# HUSKY_BLACKFLY_SECONDARY=                # default: 0
# HUSKY_BLACKFLY_SECONDARY_MOUNT_ENABLED=  # default: 1
# HUSKY_BLACKFLY_SECONDARY_MOUNT_ANGLE=    # default: 0
# HUSKY_BLACKFLY_SECONDARY_PREFIX=         # default: blackfly_secondary
# HUSKY_BLACKFLY_SECONDARY_PARENT=         # default: top_plate_link
# HUSKY_BLACKFLY_SECONDARY_XYZ=            # default: 0 0 0
# HUSKY_BLACKFLY_SECONDARY_RPY=            # default: 0 0 0

# Bumper Extension
# HUSKY_FRONT_BUMPER_EXTEND=   # default: 0
# HUSKY_REAR_BUMPER_EXTEND=    # default: 0

# Height of the sensor arch in mm.  Must be either 510 or 300
# HUSKY_SENSOR_ARCH=                # default: 0
# HUSKY_SENSOR_ARCH_HEIGHT=         # default: 510
# HUSKY_SENSOR_ARCH_OFFSET=         # default: 0 0 0
# HUSKY_SENSOR_ARCH_RPY=            # default: 0 0 0

# Extras
# ROBOT_NAMESPACE=     # /
# HUSKY_URDF_EXTRAS=   # empty.urdf
# CPR_URDF_EXTRAS=     # empty.urdf

set +a
```