husky/husky #243: Gmapping does not work

PhilipAmadasun, 2022-08-18:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 18.04)
 - ROS Distro: (e.g. Melodic)
 - Built from source or installed:
 - Package version: (if from repository, give version from `sudo dpkg -s ros-melodic-husky-navigation`, if from source, give commit hash)
 - Real hardware or simulation: Simulation

 **Expected behaviour**
 Launching gmapping_demo,visualizers in the navigation group on rviz should work.

 **Actual behaviour**
 On rviz , the navigation group has alot of status errors, and thus SLAM does not work.

**To Reproduce**
Provide the steps to reproduce:
-I simply followed the instructions in the husky ROS Wiki [here](http://wiki.ros.org/husky_navigation/Tutorials/Husky%20Gmapping%20Demo)

**Other notes**
Here is my husky.urdf.xacro file:
```
<?xml version="1.0"?>
<!--
Software License Agreement (BSD)

\file      husky.urdf.xacro
\authors   Paul Bovbel <pbovbel@clearpathrobotics.com>, Devon Ash <dash@clearpathrobotics.com>
\copyright Copyright (c) 2015, Clearpath Robotics, Inc., All rights reserved.

Redistribution and use in source and binary forms, with or without modification, are permitted provided that
the following conditions are met:
 * Redistributions of source code must retain the above copyright notice, this list of conditions and the
   following disclaimer.
 * Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the
   following disclaimer in the documentation and/or other materials provided with the distribution.
 * Neither the name of Clearpath Robotics nor the names of its contributors may be used to endorse or promote
   products derived from this software without specific prior written permission.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WAR-
RANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR
PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, IN-
DIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT
OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
-->
<robot name="husky" xmlns:xacro="http://ros.org/wiki/xacro">

  <xacro:arg name="laser_enabled" default="$(optenv HUSKY_LMS1XX_ENABLED 0)" />
  <xacro:arg name="laser_xyz" default="$(optenv HUSKY_LMS1XX_XYZ 0.2206 0.0 0.00635)" />
  <xacro:arg name="laser_rpy" default="$(optenv HUSKY_LMS1XX_RPY 0.0 0.0 0.0)" />

  <xacro:arg name="laser_secondary_enabled" default="$(optenv HUSKY_LMS1XX_SECONDARY_ENABLED 0)" />
  <xacro:arg name="laser_secondary_xyz" default="$(optenv HUSKY_LMS1XX_SECONDARY_XYZ -0.2206 0.0 0.00635)" />
  <xacro:arg name="laser_secondary_rpy" default="$(optenv HUSKY_LMS1XX_SECONDARY_RPY 0.0 0.0 3.14159)" />

  <xacro:arg name="laser_ust10_front_enabled" default="$(optenv HUSKY_UST10_ENABLED 1)" />
  <xacro:arg name="laser_ust10_front_prefix"  default="$(optenv HUSKY_UST10_PREFIX front)" />
  <xacro:arg name="laser_ust10_front_parent"  default="$(optenv HUSKY_UST10_PARENT top_plate_link)" />
  <xacro:arg name="laser_ust10_front_topic"   default="$(optenv HUSKY_UST10_TOPIC front/scan)" />
  <xacro:arg name="laser_ust10_front_xyz"     default="$(optenv HUSKY_UST10_XYZ 0.2206 0.0 0.00635)" />
  <xacro:arg name="laser_ust10_front_rpy"     default="$(optenv HUSKY_UST10_RPY 0 0 0)" />

  <xacro:arg name="laser_ust10_rear_enabled"  default="$(optenv HUSKY_UST10_SECONDARY_ENABLED 0)" />
  <xacro:arg name="laser_ust10_rear_prefix"   default="$(optenv HUSKY_UST10_SECONDARY_PREFIX rear)" />
  <xacro:arg name="laser_ust10_rear_parent"   default="$(optenv HUSKY_UST10_SECONDARY_PARENT top_plate_link)" />
  <xacro:arg name="laser_ust10_rear_topic"    default="$(optenv HUSKY_UST10_SECONDARY_TOPIC rear/scan)" />
  <xacro:arg name="laser_ust10_rear_xyz"      default="$(optenv HUSKY_UST10_SECONDARY_XYZ -0.2206 0.0 0.00635)" />
  <xacro:arg name="laser_ust10_rear_rpy"      default="$(optenv HUSKY_UST10_SECONDARY_RPY 0 0 3.14159)" />

  <xacro:arg name="laser_3d_enabled" default="$(optenv HUSKY_LASER_3D_ENABLED 0)" />
[... 231 lines cut ...]

  <!-- Optional for Clearpath internal softwares -->
  <xacro:include filename="$(arg cpr_urdf_extras)" />

</robot>

```

Here is my gmapping_demo.launch file:
```
<?xml version="1.0"?>

<launch>
  <arg name="scan_topic" default="$(eval optenv('HUSKY_LASER_TOPIC', 'scan'))" />

  <!--- Run gmapping -->
  <include file="$(find husky_navigation)/launch/gmapping.launch">
    <arg name="scan_topic" value="$(arg scan_topic)" />
  </include>

  <!--- Run Move Base -->
  <include file="$(find husky_navigation)/launch/move_base.launch" />

</launch>
```

gxfgit, 2022-08-18:
您好，邮件已经收到，我会尽快查看并回复。谢谢！

civerachb-cpr, 2022-08-18:
What are the error messages you're seeing in Rviz?  A screenshot of Rviz with the errors expanded would be helpful.  Can you run
```bash
roslaunch husky_viz view_model.launch
```
and see the Husky model rendering correctly?

When you start Gazebo are you seeing any error messages in the terminal?

Which branch did you check out when you built from source and did you install the additional dependencies by running
```bash
rosdep install --from-paths src --ignore-src -r -y
```
inside your catkin workspace folder before running `catkin_make`?

PhilipAmadasun, 2022-08-19:
@civerachb-cpr @tonybaltovski 
Yes I see the husky model rendering correclty. This is what my terminal looks like when I run gazebo( roslaunch husky_gazebo husky_playpen.launch):
```
SUMMARY
========

PARAMETERS
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: True
 * /ekf_localization/imu0_queue_size: 10
 * /ekf_localization/imu0_remove_gravitational_acceleration: True
 * /ekf_localization/odom0: husky_velocity_co...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom0_queue_size: 10
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/predict_to_current_time: True
 * /ekf_localization/two_d_mode: True
 * /ekf_localization/world_frame: odom
 * /gazebo/enable_ros_network: True
 * /husky_joint_publisher/publish_rate: 50
 * /husky_joint_publisher/type: joint_state_contr...
 * /husky_velocity_controller/angular/z/has_acceleration_limits: True
 * /husky_velocity_controller/angular/z/has_velocity_limits: True
 * /husky_velocity_controller/angular/z/max_acceleration: 6.0
 * /husky_velocity_controller/angular/z/max_velocity: 2.0
 * /husky_velocity_controller/base_frame_id: base_link
 * /husky_velocity_controller/cmd_vel_timeout: 0.25
 * /husky_velocity_controller/enable_odom_tf: False
 * /husky_velocity_controller/estimate_velocity_from_position: False
 * /husky_velocity_controller/left_wheel: ['front_left_whee...
 * /husky_velocity_controller/linear/x/has_acceleration_limits: True
 * /husky_velocity_controller/linear/x/has_velocity_limits: True
 * /husky_velocity_controller/linear/x/max_acceleration: 3.0
 * /husky_velocity_controller/linear/x/max_velocity: 1.0
 * /husky_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/publish_rate: 50
 * /husky_velocity_controller/right_wheel: ['front_right_whe...
 * /husky_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/type: diff_drive_contro...
 * /husky_velocity_controller/velocity_rolling_window_size: 2
 * /husky_velocity_controller/wheel_radius_multiplier: 1.0
 * /husky_velocity_controller/wheel_separation_multiplier: 1.875
 * /joy_teleop/joy_node/autorepeat_rate: 20
 * /joy_teleop/joy_node/deadzone: 0.1
 * /joy_teleop/joy_node/dev: /dev/input/ps4
 * /joy_teleop/teleop_twist_joy/axis_angular: 0
 * /joy_teleop/teleop_twist_joy/axis_linear: 1
 * /joy_teleop/teleop_twist_joy/enable_button: 4
 * /joy_teleop/teleop_twist_joy/enable_turbo_button: 5
 * /joy_teleop/teleop_twist_joy/scale_angular: 1.4
 * /joy_teleop/teleop_twist_joy/scale_linear: 0.4
 * /joy_teleop/teleop_twist_joy/scale_linear_turbo: 2.0
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /twist_mux/locks: [{'topic': 'e_sto...
 * /twist_mux/topics: [{'topic': 'joy_t...
 * /use_sim_time: True

NODES
  /
    base_controller_spawner (controller_manager/spawner)
    ekf_localization (robot_localization/ekf_localization_node)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
[... 91 lines cut ...]

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [5670]
[ INFO] [1660866708.155783504]: rviz version 1.13.24
[ INFO] [1660866708.155825845]: compiled against Qt version 5.9.5
[ INFO] [1660866708.155837846]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1660866708.161636304]: Forcing OpenGl version 0.
[ INFO] [1660866708.274398128]: Stereo is NOT SUPPORTED
[ INFO] [1660866708.274540067]: OpenGL device: Mesa DRI Intel(R) UHD Graphics 630 (Coffeelake 3x8 GT2) 
[ INFO] [1660866708.274614987]: OpenGl version: 3.0 (GLSL 1.3).
[ WARN] [1660866708.950775989, 1309.200000000]: Interactive marker 'robot_twist_marker' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

```
[](url
[image]
)

PhilipAmadasun, 2022-08-19:
@civerachb-cpr @tonybaltovski   I did not build any of the husky packages in a workspace. I used `sudo apt-get install ros-melodic-husky*`

civerachb-cpr, 2022-08-19:
This part of your log looks suspicious:
```
[ WARN] [1660865805.610883380, 1708.100000000]: Shutdown request received.
[ WARN] [1660865805.613278745, 1708.110000000]: Reason given for shutdown: [[/robot_state_publisher] Reason: new node registered with same name]
[robot_state_publisher-7] process has finished cleanly
```

It looks like other ROS nodes may already be running on your system; either another simulation or the nodes needed for a physical robot.  Do you have any other ROS nodes running on the computer? Or have you modified any launch files to start an additional `robot_state_publisher` node?

I just installed Ubuntu 18.04 in a VM, added the ROS and Clearpath apt + rosdep sources, and ran
```bash
sudo apt-get install ros-melodic-husky*
```
followed by
```bash
roslaunch husky_gazebo husky_playpen.launch
```

I'm seeing this output, much of which is the same as yours, but crucially without the warning about duplicate `robot_state_publisher` nodes:
```
$ roslaunch husky_gazebo husky_playpen.launch 
... logging to /home/administrator/.ros/log/450eef54-1ffa-11ed-aec0-080027d6af03/roslaunch-cpr-cib-vm-melodic-8677.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cpr-cib-vm-melodic:36449/

SUMMARY
========

PARAMETERS
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: True
 * /ekf_localization/imu0_queue_size: 10
 * /ekf_localization/imu0_remove_gravitational_acceleration: True
 * /ekf_localization/odom0: husky_velocity_co...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom0_queue_size: 10
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/predict_to_current_time: True
 * /ekf_localization/two_d_mode: True
 * /ekf_localization/world_frame: odom
 * /gazebo/enable_ros_network: True
 * /husky_joint_publisher/publish_rate: 50
 * /husky_joint_publisher/type: joint_state_contr...
 * /husky_velocity_controller/angular/z/has_acceleration_limits: True
 * /husky_velocity_controller/angular/z/has_velocity_limits: True
 * /husky_velocity_controller/angular/z/max_acceleration: 6.0
 * /husky_velocity_controller/angular/z/max_velocity: 2.0
 * /husky_velocity_controller/base_frame_id: base_link
 * /husky_velocity_controller/cmd_vel_timeout: 0.25
 * /husky_velocity_controller/enable_odom_tf: False
 * /husky_velocity_controller/estimate_velocity_from_position: False
 * /husky_velocity_controller/left_wheel: ['front_left_whee...
 * /husky_velocity_controller/linear/x/has_acceleration_limits: True
 * /husky_velocity_controller/linear/x/has_velocity_limits: True
 * /husky_velocity_controller/linear/x/max_acceleration: 3.0
 * /husky_velocity_controller/linear/x/max_velocity: 1.0
 * /husky_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/publish_rate: 50
 * /husky_velocity_controller/right_wheel: ['front_right_whe...
 * /husky_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/type: diff_drive_contro...
 * /husky_velocity_controller/velocity_rolling_window_size: 2
 * /husky_velocity_controller/wheel_radius_multiplier: 1.0
[... 95 lines cut ...]
		wheel separation multiplier: 1.875
	Publication parameters:
		Publish executed velocity command: disabled
		Publication rate: 50
		Publish frame odom on tf: disabled
```

About a few of the errors and warnings:

```
[ERROR] [1660939531.029559968]: Couldn't open joystick /dev/input/ps4. Will retry every second.
```
This is normal; I don't have a PS4 controller connected, so the default teleop input will throw this warning.  You can override the default controller with by setting
```
export HUSKY_LOGITECH=1
export HUSKY_JOY_DEVICE=/dev/input/js0  # substitute as appropriate
```

```
[ERROR] [1660939536.010355249, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/front_left_wheel
[ERROR] [1660939536.037363580, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/front_right_wheel
[ERROR] [1660939536.068640007, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/rear_left_wheel
[ERROR] [1660939536.089464244, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/rear_right_wheel
```
This is a long-standing issue with Gazebo. The Husky's wheels do not use PID control, so no PID gains are specified.  This shouldn't be reported as an error, but Gazebo does anyway.

tonybaltovski, 2022-10-14:
Closing due to inactivity.  Please feel free to re-open if issue persists.

closed 2022-10-14 by tonybaltovski

w3ichen, 2024-06-08:
I'm getting an error of `Thread 22 "gzserver" received signal SIGILL, Illegal instruction`, when running gmapping_demo.launch.

**System**
- Apple Silicon M2
- UTM Virtual Machine Ubuntu 20.04 x86_64 emulation
- Ros1 noetic
- Gazebo11

**Reproduction**
1. Run `roslaunch husky_gazebo husky_playpen.launch`, no errors at this point
2. Run `roslaunch husky_viz view_robot.launch` will cause gazebo to error out

**Error**
 ```bash
Illegal instruction (core dumped)
[gazebo-2] process has died [pid 13647, exit code 132, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/husky
name: =gazebo
ot/src/mobile_manipulator/worlds/env.world __name:=gazebo - 1og:=/home/.ros/log/8ba903b4-242b-11ef-bc74-bbef25f09167/gazebo-2.log].
log file: /home/.ros/log/8ba903b4-242b-11ef-bc74-bbef25f09167/gazebo-2*.
```

GDB backtrace is showing gazebo math function is failing:
```bash
(gdb) bt

   #0  0x00007ffff60d6963 in double std::generate_canonical<double, 53ul, std::mersenne_twister_engine<unsigned long, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul> >(std::mersenne_twister_engine<unsigned long, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>&)
    () at /lib/x86_64-linux-gnu/libignition-math6.so.6
#1  0x00007ffff60d6a04 in double std::normal_distribution<double>::operator()<std::mersenne_twister_engine<unsigned long, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul> >(std::mersenne_twister_engine<unsigned long, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>&, std::normal_distribution<double>::param_type const&) ()
    at /lib/x86_64-linux-gnu/libignition-math6.so.6
#2  0x00007ffff60d6154 in ignition::math::v6::Rand::DblNormal(double, double)
    () at /lib/x86_64-linux-gnu/libignition-math6.so.6
#3  0x00007ffff72b16e7 in gazebo::sensors::GaussianNoiseModel::ApplyImpl(double, double) () at /lib/x86_64-linux-gnu/libgazebo_sensors.so.11
#4  0x00007ffff72cdf71 in gazebo::sensors::Noise::Apply(double, double) ()
    at /lib/x86_64-linux-gnu/libgazebo_sensors.so.11
#5  0x00007ffff72d1286 in gazebo::sensors::RaySensor::UpdateImpl(bool) ()
    at /lib/x86_64-linux-gnu/libgazebo_sensors.so.11
#6  0x00007ffff72d59a9 in gazebo::sensors::Sensor::Update(bool) ()
    at /lib/x86_64-linux-gnu/libgazebo_sensors.so.11
#7  0x00007ffff72e0f35 in gazebo::sensors::SensorManager::SensorContainer::Update(bool) () at /lib/x86_64-linux-gnu/libgazebo_sensors.so.11
#8  0x00007ffff72e0797 in gazebo::sensors::SensorManager::SensorContainer::RunLoop() () at /lib/x86_64-linux-gnu/libgazebo_sensors.so.11
#9  0x00007ffff59b543b in  ()
    at /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#10 0x00007ffff6f28609 in start_thread (arg=<optimized out>)
    at pthread_create.c:477
#11 0x00007ffff7464353 in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```