husky/husky #253: Invalid parameter robot_description

Karin-Sugi, 2022-10-14:
Hi, I'm making customized husky , and tryied to show it following [husky tutorial](http://www.clearpathrobotics.com/assets/guides/melodic/husky/CustomizeHuskyConfig.html). 

When I try to see it on rviz, rviz is launched, but the model isn't shown("robot model" turn to red cahracters)! On gazebo, gazebo isn't launched and error below is returned.

commands i send:
`
$ export HUSKY_URDF_EXTRAS=~/path/to/t/customized/file.urdf.xacro
$ roslaunch husky_gazebo husky_empty_world.launch 
($ roslaunch husky_viz view_robot.launch)
`

returned error when i launch gazebo:
`RLException: while processing /opt/ros/melodic/share/husky_gazebo/launch/spawn_husky.launch:
while processing /opt/ros/melodic/share/husky_control/launch/control.launch:
while processing /opt/ros/melodic/share/husky_description/launch/description.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/opt/ros/melodic/share/husky_description/urdf/husky.urdf.xacro', 'robot_namespace:=/']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro '$(find husky_description)/urdf/husky.urdf.xacro'     robot_namespace:=$(arg robot_namespace)" name="robot_description"/>
The traceback for the exception was written to the log file
`

I'm using ros Melodic, and LMS1xx package has been downloaded. Also, my urdf can be seen on rviz if I try it using `roslaunch urdf_tutorial display.launch model:=hauser.urdf.xacro`,so urdf itself must be fine.

How can I solve it?

Thanks in advance,
Karin

civerachb-cpr, 2022-10-14:
What are the contents of the URDF file you're assigning here:

```
export HUSKY_URDF_EXTRAS=~/path/to/the/customized/file.urdf.xacro
```

and are you properly replacing the path with the actual path to your URDF file?

Karin-Sugi, 2022-10-17:
Hi, the path is fine but my urdf is not.
I said it must be fine because `roslaunch urdf_tutorial display.launch model:=hauser.urdf.xacro` showed my model with no problem. However, when I use husky's default lanuch file to launch rviz & gazebo, it seems I shouldn't have used `<xacro:include  filename:"path/to/the/original_husky.urdf.xacro">`. Once, i remove the line, both rviz&gazebo show my model correctly!!

closed 2022-10-17 by Karin-Sugi