husky/husky #256: JoyStick cannot notrol husky

Karin-Sugi, 2022-10-21:
Hi, as i mentioned in the title, my husky bacame unable to be cobtroled by joystick suddenly!

All lights (battery, comm...) are green, but husky never move. 
I sent command by ssh to both `/husky_velocity_controller/cmd_vel` and `/joy_teleop/cmd_vel`. Both of them worked(/joy_teleop/cmd_vel didn't worked at first but second time, fine). Also, i tried `rostopic echo /joy_teleop/cmd_vel` while using joy sticks, and it returns ONLY one data as like below.
```
linear:
        x: 0.0
        y: 0.0
        z: 0.0
angular:
        x: 0.0
        y: 0.0
        z: 0.0
```
I think the data from joy stick isn't sent to the ros topic, but it seems the USB of joy stick is recognized (I saw its name when I do lsusb).

How can i solve it?
I use ros-melodic and my husky is "HUSKY A200".

Thanks in advance,
Karin

civerachb-cpr, 2022-10-21:
What brand of controller are you using?  Is it a PS4 controller, or a Logitech F710?  (I'm assuming it's an F710 since you mentioned it has a USB dongle, but it's good to confirm.)

What is the output of the following commands on your robot?
```
ls -l /dev/input
rostopic echo /joy_teleop/joy -n 1
```

Karin-Sugi, 2022-10-25:
Hi, 
It is surprising suddenly husky starts moving with joy stick. Now I can use it, but I'm not sure why it didn't work and why it starts to work again. If you have any idea, could you tell me in the case it stop moving again.

I'm using logitech F710.  and the resutls of the commands were below.

```
$ ls -l /dev/input

total 0
drwxr-xr-x 2 root root      80 Oct 25 01:18 by-id
drwxr-xr-x 2 root root      80 Oct 25 01:18 by-path
lrwxrwxrwx 1 root root       3 Oct 25 01:18 ds4x -> js1
crw-rw-rw- 1 root input 13, 64 Oct 25 01:18 event0
crw-rw-rw- 1 root input 13, 65 Oct 25 01:18 event1
crw-rw-rw- 1 root input 13, 74 Oct 25 01:18 event10
crw-rw-rw- 1 root input 13, 75 Oct 25 01:18 event11
crw-rw-rw- 1 root input 13, 76 Oct 25 01:18 event12
crw-rw-rw- 1 root input 13, 66 Oct 25 01:18 event2
crw-rw-rw- 1 root input 13, 67 Oct 25 01:18 event3
crw-rw-rw- 1 root input 13, 68 Oct 25 01:18 event4
crw-rw-rw- 1 root input 13, 69 Oct 25 01:18 event5
crw-rw-rw- 1 root input 13, 70 Oct 25 01:18 event6
crw-rw-rw- 1 root input 13, 71 Oct 25 01:18 event7
crw-rw-rw- 1 root input 13, 72 Oct 25 01:18 event8
crw-rw-rw- 1 root input 13, 73 Oct 25 01:18 event9
lrwxrwxrwx 1 root root       3 Oct 25 01:18 f710 -> js0
crw-rw-r-- 1 root input 13,  0 Oct 25 01:18 js0
crw-rw-r-- 1 root input 13,  1 Oct 25 01:18 js1
crw-rw---- 1 root input 13, 63 Oct 25 01:18 mice
lrwxrwxrwx 1 root root       3 Oct 25 01:18 ps4 -> js1
```

```
$ rostopic echo /joy_teleop/joy -n 1

header: 
  seq: 18692
  stamp: 
    secs: 1666676059
    nsecs: 510209154
  frame_id: "/dev/input/js0"
axes: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
buttons: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
---
```

civerachb-cpr, 2022-10-25:
The issue is likely caused by the `ds4drv` service; this service creates a virtual `/dev/input/js*` device that manages the PS4 controller.  (The default driver for the PS4 controller can latch the last command if the robot drives out of range; the ds4drv service handles safely disconnecting the input if it goes out of range.)

Depending on the relative order of when the `dsrdrv` service started and when the USB dongle was detected, it's possible that `/dev/input/js0` and `/dev/input/js1` were inverted.  Because your teleop_joy node is listening to /dev/input/js0 it may have been listening to the virtual PS4 input by mistake.

The best way to avoid this happening in the future is:

1) Disable the ds4drv service; you aren't using a PS4 controller, so this service is not necessary.  Run `sudo systemctl disable ds4drv` should prevent it from starting up again.

2) Edit `/etc/udev/rules.d` and make sure the following envars are set:
```
export HUSKY_LOGITECH=1
export HUSKY_JOY_DEVICE=/dev/input/f710
```
This will reconfigure the teleop_joy node to read from /dev/input/f710, which will be a symbolic link pointed to the jsN device corresponding to your game controller.

civerachb-cpr, 2022-10-26:
I'm going to mark this as closed, but if this issue persists please re-open it or contact our support team directly.

closed 2022-10-26 by civerachb-cpr

Karin-Sugi, 2023-06-05:
Hi, I've re-opened this thread because same issue had occured again!
Last time, without any operation, joy stick started working, so I did nothing, actually.

This time, I followed your suggestion. After disable ds4drv.  I found `/etc/udev/rules.d` is directry, where I couldn't find `export .....` lines you mentioned before. How and where can I add this lines? Only disabling the ds4drv, my joy didn't work well. 

Also, this time, wired joy stick can control husky. It is very strange for me because logitech joy with dongle doesn't work when wired logicool joy works fine.

After disable ds4drv, the results of command `ls -l /dev/input` is below.
```
total 0
drwxr-xr-x 2 root root      80 Jun  5 03:31 by-id
drwxr-xr-x 2 root root      80 Jun  5 03:31 by-path
crw-rw-rw- 1 root input 13, 64 Jun  5 03:31 event0
crw-rw-rw- 1 root input 13, 65 Jun  5 03:31 event1
crw-rw-rw- 1 root input 13, 74 Jun  5 03:31 event10
crw-rw-rw- 1 root input 13, 75 Jun  5 03:31 event11
crw-rw-rw- 1 root input 13, 66 Jun  5 03:31 event2
crw-rw-rw- 1 root input 13, 67 Jun  5 03:31 event3
crw-rw-rw- 1 root input 13, 68 Jun  5 03:31 event4
crw-rw-rw- 1 root input 13, 69 Jun  5 03:31 event5
crw-rw-rw- 1 root input 13, 70 Jun  5 03:31 event6
crw-rw-rw- 1 root input 13, 71 Jun  5 03:31 event7
crw-rw-rw- 1 root input 13, 72 Jun  5 03:31 event8
crw-rw-rw- 1 root input 13, 73 Jun  5 03:31 event9
lrwxrwxrwx 1 root root       3 Jun  5 03:31 f310 -> js0
crw-rw-r-- 1 root input 13,  0 Jun  5 03:31 js0
crw-rw---- 1 root input 13, 63 Jun  5 03:31 mice
```

Also, files in my `/etc/udev/rules.d` directry is below
```
41-clearpath.rules  41-xbox.rules  60-startech.rules
41-logitech.rules   52-ftdi.rules
```

Thanks in advance,
Karin

civerachb-cpr, 2023-06-06:
The `export ...` lines should go in `/etc/ros/setup.bash`.  It looks like you're using a Logitech F310, not a PS4 controller.  The `ds4drv` package is only needed for the PS4 controllers, and can be disabled/uninstalled if you don't need it.

To use the logitech controller you should be able to add these lines to the end of `/etc/ros/setup.bash`:

```
export HUSKY_LOGITECH=true
export HUSKY_JOY_DEVICE=/dev/input/f310
```

and then either reboot the robot or run `sudo systemctl restart ros`

Karin-Sugi, 2023-06-12:
I could. Thanks!