husky/husky #263: Cant find /dev/clearpath/gps file

Rowaid-x, 2022-10-31:
- OS: Ubuntu 18.04
 - ROS Melodic

I just installed husky software(melodic) and when done tried to run "roslaunch husky_bringup navsat.launch"

Error :
Couldn't open dev/clearpath/gps

Same thing for imu

tonybaltovski, 2022-10-31:
Do you have those devices?  Can you see them using `lsub`?  How did you install the Husky software?

Rowaid-x, 2022-10-31:
I have them yes, I installed it using the instructions [here](https://www.clearpathrobotics.com/assets/guides/kinetic/husky/InstallHuskySoftware.html)

I just chose the melodic version, connected my ps4 controller. 

Only the laser is available in the rostopic list.

I will try tomorrow the lsusb but am pretty sure they're connected.

Rowaid-x, 2022-10-31:
Does the usb port which the um7 is connected to matters?

tonybaltovski, 2022-10-31:
Did you run `rosrun husky_bringup install` with them connected?  The port does not matter.

Rowaid-x, 2022-10-31:
Yes ,and it didn't work

tonybaltovski, 2022-11-01:
Can you get the `lsusb' and `dmesg` output?

Rowaid-x, 2022-11-02:
The um7 is detected it was a loose connection sorry for that but gps no,

Dmesg output:
'''
[    0.000000] microcode: microcode updated early to revision 0xf0, date = 2021-11-15
[    0.000000] Linux version 4.15.0-194-generic (buildd@lcy02-amd64-052) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #205-Ubuntu SMP Fri Sep 16 19:49:27 UTC 2022 (Ubuntu 4.15.0-194.205-generic 4.15.18)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-194-generic root=/dev/mapper/cpr--unassigned--vg-root ro biosdevname=0 ipv6.disable=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000005efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000005f000-0x000000000005ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000060000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000008bd39fff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bd3a000-0x000000008c19ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008c1a0000-0x000000008c21cfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008c21d000-0x000000008c664fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008c665000-0x000000008cd0efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008cd0f000-0x000000008cd0ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008cd10000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x82deb018-0x82dfb057] usable ==> usable
[    0.000000] e820: update [mem 0x82deb018-0x82dfb057] usable ==> usable
[    0.000000] e820: update [mem 0x82dda018-0x82deac57] usable ==> usable
[    0.000000] e820: update [mem 0x82dda018-0x82deac57] usable ==> usable
[    0.000000] e820: update [mem 0x82dc9018-0x82dd9e57] usable ==> usable
[    0.000000] e820: update [mem 0x82dc9018-0x82dd9e57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000005efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000005f000-0x000000000005ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000060000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000082dc9017] usable
[    0.000000] reserve setup_data: [mem 0x0000000082dc9018-0x0000000082dd9e57] usable
[    0.000000] reserve setup_data: [mem 0x0000000082dd9e58-0x0000000082dda017] usable
[    0.000000] reserve setup_data: [mem 0x0000000082dda018-0x0000000082deac57] usable
[    0.000000] reserve setup_data: [mem 0x0000000082deac58-0x0000000082deb017] usable
[    0.000000] reserve setup_data: [mem 0x0000000082deb018-0x0000000082dfb057] usable
[    0.000000] reserve setup_data: [mem 0x0000000082dfb058-0x000000008bd39fff] usable
[    0.000000] reserve setup_data: [mem 0x000000008bd3a000-0x000000008c19ffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008c1a0000-0x000000008c21cfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000008c21d000-0x000000008c664fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000008c665000-0x000000008cd0efff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008cd0f000-0x000000008cd0ffff] usable
[    0.000000] reserve setup_data: [mem 0x000000008cd10000-0x000000008fffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[... 870 lines cut ...]
[ 2937.455169] usb 1-3: new high-speed USB device number 7 using xhci_hcd
[... cut ...]
[ 2970.480903] br0: port 2(enp1s0) entered forwarding state
administrator@cpr-a200-0808:/$

'''

Lsusb
Output:
'''

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 007: ID 03e7:2150  
Bus 001 Device 005: ID 8087:0025 Intel Corp. 
Bus 001 Device 002: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
'''

tonybaltovski, 2022-11-02:
Not seeing a GPS device there.  Which GPS is it and how is it connected?

Rowaid-x, 2022-11-02:
Swiftnav duro ,its connected by ethernet cable

tonybaltovski, 2022-11-02:
Could I get you to email support@clearpathrobotics.com to provide you with detailed instructions to get the Swiftnav Duro working again.

closed 2022-11-25 by tonybaltovski