husky/husky #264: Um7 packet without data

Rowaid-x, 2022-11-02:
**Please provide the following information:**
 - OS: 18.04
 - ROS Distro: Melodic
 - Real hardware 

Um7 doesn't receive data

Rosrun um7 um7_driver

rosrun um7 um7_driver 
[ INFO] [1667367895.042224127]: um7_driver successfully connected to serial port /dev/clearpath/um7.
[ INFO] [1667367895.065160976]: Received packet 02 without data.
[ INFO] [1667367895.068245150]: Setting update rate to 160Hz
[ INFO] [1667367895.092220604]: Received packet 02 without data.
[ INFO] [1667367895.342022190]: Received packet 04 without data.
[ERROR] [1667367895.895289034]: Unable to set CREG_COM_RATES5.
[ INFO] [1667367895.895379002]: Attempting reconnection after error.
[ INFO] [1667367896.896302190]: um7_driver successfully connected to serial port /dev/clearpath/um7.
[ INFO] [1667367896.926591566]: Received packet 02 without data.
[ INFO] [1667367896.935063056]: Setting update rate to 160Hz
[ INFO] [1667367896.938572218]: Received packet 02 without data.
[ INFO] [1667367896.941582882]: Received packet 04 without data.
[ INFO] [1667367896.944535374]: Received packet 05 without data.
[ INFO] [1667367896.947561436]: Received packet 06 without data.
[ WARN] [1667367896.955753342]: Excluding magnetometer updates from EKF.
[ INFO] [1667367896.975545165]: Received packet 08 without data.
[ INFO] [1667367896.979696600]: Sending command: zero gyroscopes
[ INFO] [1667367896.987522327]: Received packet ad without data.

Tried also changing the update rate with no difference.

Any suggestions?

tonybaltovski, 2022-11-02:
Did you change the update rate for this device?

Rowaid-x, 2022-11-02:
Yes tried changing multiple times

tonybaltovski, 2022-11-02:
Does the issue happen when you are operating at 20Hz?

closed 2022-11-25 by tonybaltovski