husky/husky #279: Vizualizing husky in Rviz from remote computer

orjano-max, 2023-03-13:
**Please provide the following information:**
 - OS: Jetpack 5.0.2, Nvidia AGX Xavier
 - ROS Distro: Galactic
 - Built from source or installed: built from source
 - Real hardware or simulation: real

 **Expected behaviour**
I am running a clearpath husky with packages built from source on an xavier. I can launch the husky on the xavier and everything works fine.

I want to be able to view the RobotModel on rviz on my remote computer. Usually i am launching the husky through ssh.
 
 **Actual behaviour**
 I can bring up rviz2 and view the robot model by selecting RobotModel and using "robot_description" as description topic.
However, when trying to view the robot on my remote computer, i add "RobotModel" in rviz and set "robot_description" as description topic. The robot model seems fine in rviz2  on the toolbar, but nothing shows up in the workspace.
I get this output to the terminal, but for all the links:
```bash
[ERROR] [1678715871.373767837] [rviz2]: could not load model 'file:///home/husky/Git/ORJANOS_MASTER_REPO/install/husky_description/share/husky_description/meshes/user_rail.dae' for link 'user_rail_link': FileNotFoundException: Cannot locate resource file:///home/husky/Git/ORJANOS_MASTER_REPO/install/husky_description/share/husky_description/meshes/user_rail.dae in resource group OgreAutodetect. in ResourceGroupManager::openResource at /tmp/binarydeb/ros-galactic-rviz-ogre-vendor-8.5.2/.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 703)
```

From the message above, it seems to try to open the file from the local xavier. Btw. meshes is added to the share folder. I want to emphasize that this works fine when trying to view the robot locally on the husky.

Is it possible to get the model to show up when using rviz2 on a remote computer. I have sourced the packages before launching rviz2 so the "husky_description" package should be available on the remote computer.

tonybaltovski, 2023-03-13:
Can you try this [branch](https://github.com/husky/husky/tree/mesh-fix)?  Also, we aren't really supporting Galactic, especially since it is EOL.

orjano-max, 2023-03-13:
Yes i know galactic is eol, but sadly, i cannot use humble as Xavier does not support ubuntu 22 yet. And i would rather use Foxy, but due to compatibility issues with an interbotix vx300 robotic arm and Foxy, i will have to stick to galactic.
I am thankful for the help even though galactic is eol.

It seems that the branch is not building. From the error message, it seems like there is something wrong this the call to CallBackReturn. Is this function different from the one it has inherited from? Because i did actually use this call for CallBackReturn in the source code from the galactic branch:  `rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn`. I should probably have reported this bug earlier too but i don't really know if my quickfix is sufficient.

cloned git repo
sourced ros galactic
rosdep install --from-path src --ignore-src --rosdistro galactic -y
colcon build
got this build error:

```bash
--- stderr: husky_base                              
In file included from /home/orjan/husky_mes_fix_ws/src/husky/husky_base/src/husky_hardware.cpp:32:
/home/orjan/husky_mes_fix_ws/src/husky/husky_base/include/husky_base/husky_hardware.hpp:32:23: error: ‘CallbackReturn’ in namespace ‘hardware_interface’ does not name a type
   32 |   hardware_interface::CallbackReturn on_init(const hardware_interface::HardwareInfo & info) override;
      |                       ^~~~~~~~~~~~~~
/home/orjan/husky_mes_fix_ws/src/husky/husky_base/include/husky_base/husky_hardware.hpp:41:23: error: ‘CallbackReturn’ in namespace ‘hardware_interface’ does not name a type
   41 |   hardware_interface::CallbackReturn on_activate(const rclcpp_lifecycle::State & previous_state) override;
      |                       ^~~~~~~~~~~~~~
/home/orjan/husky_mes_fix_ws/src/husky/husky_base/include/husky_base/husky_hardware.hpp:44:23: error: ‘CallbackReturn’ in namespace ‘hardware_interface’ does not name a type
   44 |   hardware_interface::CallbackReturn on_deactivate(const rclcpp_lifecycle::State & previous_state) override;
      |                       ^~~~~~~~~~~~~~
/home/orjan/husky_mes_fix_ws/src/husky/husky_base/src/husky_hardware.cpp:194:21: error: ‘CallbackReturn’ in namespace ‘hardware_interface’ does not name a type
  194 | hardware_interface::CallbackReturn HuskyHardware::on_init(const hardware_interface::HardwareInfo & info)
      |                     ^~~~~~~~~~~~~~
/home/orjan/husky_mes_fix_ws/src/husky/husky_base/src/husky_hardware.cpp:313:21: error: ‘CallbackReturn’ in namespace ‘hardware_interface’ does not name a type
  313 | hardware_interface::CallbackReturn HuskyHardware::on_activate(const rclcpp_lifecycle::State & /*previous_state*/)
      |                     ^~~~~~~~~~~~~~
/home/orjan/husky_mes_fix_ws/src/husky/husky_base/src/husky_hardware.cpp:332:21: error: ‘CallbackReturn’ in namespace ‘hardware_interface’ does not name a type
  332 | hardware_interface::CallbackReturn HuskyHardware::on_deactivate(const rclcpp_lifecycle::State & /*previous_state*/)
      |                     ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/husky_hardware.dir/build.make:63: CMakeFiles/husky_hardware.dir/src/husky_hardware.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/husky_hardware.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< husky_base [7.13s, exited with code 2]
```

tonybaltovski, 2023-03-13:
How did you have it build before?  Can you cherry-pick https://github.com/husky/husky/commit/e7693cba9e2a5b55c298e7fa0a9c2230e95a40f2?

orjano-max, 2023-03-13:
I actually copied the packages from the galactic branch to my own repo and did some changes described in my issue [here](https://github.com/orjano-max/ORJANOS_MASTER_REPO/issues/9).

I have added the fix from your commit to the files in my own repo [here](https://github.com/orjano-max/ORJANOS_MASTER_REPO/tree/galactic/code/src). I will test it tomorrow when i have access to the hardware and report back to you.

orjano-max, 2023-03-14:
It worked! Changing to package path fixed the issue regarding viewing the robot on the remote computer!
However, it seems like there are some issues with gazebo regarding using the package path. Maybe a solution wold be to have the path as file path if gazebo sim is launched.

Also, as i said, i have done some modifications to the `husky_base` in order to get them to build on galactic. I am not sure if it is sufficient but it seems to work well. I have also added `interactive_marker_twist_server` to my [repo](https://github.com/orjano-max/ORJANOS_MASTER_REPO/tree/galactic/code/src) in order to use similar teleop packages as those used for foxy and humble. It is therefore large changes in the `husky_control` package related to this teleop modification.

Anyways, thanks for the help!

closed 2023-03-16 by tonybaltovski