husky/husky #282: Humble Migration

iliabaranov, 2023-03-28:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: build from source
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-husky-PACKAGE_WITH_ISSUE`, if from source, give commit hash)
 - Real hardware or simulation: Real

 **Expected behaviour**
Hey @tonybaltovski ! :) 
I have a client who would like to test on a Husky, using the humble branch.

 **Actual behaviour**
I want to build your humble branch, however I get the error:
In file included from /home/robot/husky/src/husky/husky_base/src/husky_hardware.cpp:32:                        
/home/robot/husky/src/husky/husky_base/include/husky_base/husky_hardware.hpp:8:10: fatal error: hardware_interface/base_interface.hpp: No such file or directory                                                              
    8 | #include "hardware_interface/base_interface.hpp"   
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                       
compilation terminated.     

Seems that the this needs to be updated asap with the changes listed here:
https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/hardware_components_userdoc.html

I'm going to attempt to change this manually, but we'll see if this fixes it for us.

**To Reproduce**
Provide the steps to reproduce:
Follow stadard ROS2 Humble install.
Try to build husky base packages

**Other notes**
Hello Clearpath Team

adeeb10abbas, 2023-03-29:
I can second this. I am running into the same issue. Thanks!

tonybaltovski, 2023-03-29:
Hi @iliabaranov,

There isn't official support Humble on the Husky yet since there are some changes we are looking to implement.  The Humble branch was created for this PR: https://github.com/husky/husky/pull/275.  Try that and it should work for you.

tonybaltovski, 2024-01-02:
For ROS 2 Humble, this repository is no longer used. Please visit [clearpath_common](https://github.com/clearpathrobotics/clearpath_common). Check the [documentation](https://docs.clearpathrobotics.com/docs/ros/) for more details including supported sensors.

closed 2024-01-02 by tonybaltovski