husky/husky #283: There is no information about lateral velocity in odom topics (Gazebo)

MuhammadAlhaddad, 2023-04-02:
Good afternoon / Good evening!
I think if the robot is of the skid-steering type, then there should be an instantaneous center of rotation (ICR) which is determined based on the value of velocity on the y-axis in the local frame. However, during my examination of the topics that give information about position/velocities, I found that the velocity on the y-axis is always equal to zero, even though the robot was moving in a circular motion. I have looked at the both published topics` /husky_velocity_controller/odom` and `/odometry/filtered` and the results are illustrated in pictures. 
Question is: Why is there no value of velocity on y-axis? If it can be read in other topics then which are these topics?
Thanks in advance.
[image]
[image]

civerachb-cpr, 2023-04-03:
The Husky is not capable of driving laterally, which is why the lateral velocity is published as zero.  The robot's center of rotation should have linear velocity measured along the X axis (to indicate forward/reverse velocity) and angular velocity around the Z axis (to indicate rotation).  Any lateral velocity is unintended; the robot is not producing lateral motion as a result of the wheels, and there are no sensors in the drive system to measure lateral motion.

If you're finding that the skid-steering nature of the Husky is introducing too much nose into the odometry estimate, I would advice adding an IMU and modifying the EKF node to take the IMU data (which will include linear acceleration in all directions) as inputs into the odometry estimate.

MuhammadAlhaddad, 2023-04-04:
@civerachb-cpr  Thanks for answer. Actually, four-wheels mobile robot like Husky is of type skid-steering mobile robot in which the center of rotation in circular motion is not located on wheels axis so there should be a value for the lateral velocity in local frame. In article (_Kozłowski, Krzysztof, and Dariusz Pazderski. "Modeling and control of a 4-wheel skid-steering mobile robot." International journal of applied mathematics and computer science 14.4 (2004): 477-496._) it is written that `In contrast to most wheeled vehicles, the lateral velocity of the skid-steering mobile robot , vy , is generally nonzero. ` The figure below shows the velocities of these robot in the mentioned work 
[image]

civerachb-cpr, 2023-04-04:
In the diagram above the CoM is located closer to one pair of wheels than the other.  In the case where the CoM is equidistant from all 4 wheels the result should be that `v_y` is zero, correct?  Assuming `V_1_y = -V_2_y` and `V_3_y = -V_4_y` then the lateral velocities on the equidistant CoM will all cancel out, resulting in zero lateral velocity at that point.

For the purposes of the Husky simulation, we assume that the CoM is equidistant from all 4 wheels, or close enough to this point that the error is negligible.  We can potentially re-evaluate these assumptions at a later date.

In practical terms, is the lack of lateral velocity information actually causing errors or hindering your research?  If they are actively causing problems please feel free to submit a merge request with the necessary changes and we'll be happy to review it.

tonybaltovski, 2023-04-19:
Closing due to inactivity.

closed 2023-04-19 by tonybaltovski