husky/husky #290: not find the husky_controller

ambglh, 2023-05-23:
My ROS version is Noetic, and I am using Husky for learning purposes. I have customized the configuration of the Husky robot and converted its description files to URDF format. However, I have noticed that there are no controllers available for Husky after the conversion to URDF. As a result, my robot is unable to move. Could you please advise me on how to resolve this issue?

ambglh, 2023-05-23:
I am working on a simulation project using the Noetic branch. I have downloaded the branch and made modifications to the `husky.urdf.xacro` file to add the sensors I need. As a result, I have generated the URDF file, and the model can be displayed in Gazebo. However, there are no controllers and no `cmd_vel` topic for velocity control. Here is the URDF file I have generated.
<?xml version="1.0" ?>

<robot name="husky">
  
  <material name="dark_grey">
    <color rgba="0.2 0.2 0.2 1.0"/>
  </material>
  <material name="medium_grey">
    <color rgba="0.6 0.6 0.6 1.0"/>
  </material>
  <material name="light_grey">
    <color rgba="0.8 0.8 0.8 1.0"/>
  </material>
  <material name="yellow">
    <color rgba="0.8 0.8 0.0 1.0"/>
  </material>
  <material name="black">
    <color rgba="0.15 0.15 0.15 1.0"/>
  </material>
  <material name="white">
    <color rgba="1.0 1.0 1.0 1.0"/>
  </material>
  <material name="red">
    <color rgba="1.0 0.0 0.0 1.0"/>
  </material>
  
  <link name="base_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://husky_description/meshes/base_link.dae"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0.0 0 0.061875"/>
      <geometry>
        <box size="2.9874 0.5709 0.12375"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0.17562499999999998"/>
      <geometry>
        <box size="0.7899200000000001 0.5709 0.10375"/>
      </geometry>
    </collision>
  </link>
  
  <link name="base_footprint"/>
  <joint name="base_footprint_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 -0.13228"/>
    <parent link="base_link"/>
    <child link="base_footprint"/>
  </joint>
  
  <link name="inertial_link">
    <inertial>
      <mass value="46.034"/>
      <origin xyz="-0.00065 -0.085 0.062"/>
      <inertia ixx="0.6022" ixy="-0.02364" ixz="-0.1197" iyy="1.7386" iyz="-0.001544" izz="2.0296"/>
    </inertial>
  </link>
  <joint name="inertial_joint" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="base_link"/>
    <child link="inertial_link"/>
  </joint>
  <link name="front_left_wheel_link">
    <inertial>
      <mass value="2.637"/>
[... 383 lines cut ...]
      <rateDrift>0.005 0.005 0.005 </rateDrift>
      <rateGaussianNoise>0.005 0.005 0.005 </rateGaussianNoise>
      <headingDrift>0.005</headingDrift>
      <headingGaussianNoise>0.005</headingGaussianNoise>
    </plugin>
  </gazebo>
  <gazebo>
    <plugin filename="libhector_gazebo_ros_gps.so" name="gps_controller">
      <robotNamespace>/</robotNamespace>
      <updateRate>40</updateRate>
      <bodyName>base_link</bodyName>
      <frameId>base_link</frameId>
      <topicName>navsat/fix</topicName>
      <velocityTopicName>navsat/vel</velocityTopicName>
      <referenceLatitude>49.9</referenceLatitude>
      <referenceLongitude>8.9</referenceLongitude>
      <referenceHeading>90</referenceHeading>
      <referenceAltitude>0</referenceAltitude>
      <drift>0.0001 0.0001 0.0001</drift>
    </plugin>
  </gazebo>
  
  
  
</robot>

ambglh, 2023-05-24:
I run roslaunch husky_gazebo husky_playpen.launch does not report an error, but I want to define a new husky by myself, for example, add a new robotic arm based on it, how should this be done.<joint name="rear_left_wheel" type="continuous">
    <parent link="base_link"/>
    <child link="rear_left_wheel_link"/>
    <origin rpy="0 0 0" xyz="-0.256 0.2854 0.03282"/>
    <axis rpy="0 0 0" xyz="0 1 0"/>
  </joint>
  **<transmission name="rear_left_wheel_trans" **type="SimpleTransmission">**
    <type>transmission_interface/SimpleTransmission</type>
    <actuator name="rear_left_wheel_motor">
      <mechanicalReduction>1</mechanicalReduction>**
    </actuator>
    <joint name="rear_left_wheel">
      <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </joint>
    This is the urdf file I generated through husky.urdf.xacro, you can see the library file without the controller after conversion.
    -----
And I saw this in the srf model file, my car can't move now, I think it's because of this reason, how can I find this library file
**<plugin** name="husky_diff_controller" **filename="libhusky_gazebo_plugins.so">**
      <alwaysOn>true</alwaysOn>
      <updateRate>100.0</updateRate>
      <backLeftJoint>back_left_joint</backLeftJoint>
      <backRightJoint>back_right_joint</backRightJoint>
      <frontLeftJoint>front_left_joint</frontLeftJoint>
      <frontRightJoint>front_right_joint</frontRightJoint>
      <wheelSeparation>0.5709</wheelSeparation>
And I saw this in the srf model file, my car can't move now, I think it's because of this reason, how can I find this library file

civerachb-cpr, 2023-05-24:
The recommended way to add an arm to the Husky is to use the HUSKY_URDF_EXTRAS environment variable.  This environment variable points to a URDF file that gets included in the Husky model to add additional joints and links.

For example, the [husky_manipulation](https://github.com/husky/husky_manipulation) repository has this example to add a Kinova Gen3 arm to the Husky:

`husky_kinova_description/urdf/husky_gen3_description.urdf.xacro`
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:arg name="arm" default="gen3" />
  <xacro:arg name="dof" default="7" />
  <xacro:arg name="vision" default="true" />

  <xacro:arg name="gripper" default="" />

  <xacro:arg name="sim" default="false" />
  <xacro:arg name="prefix" default="kinova_arm_" />

  <xacro:include filename="$(find kortex_description)/robots/kortex_robot.xacro" />
  <xacro:load_robot arm="$(arg arm)" gripper="$(arg gripper)" dof="$(arg dof)" vision="$(arg vision)" sim="$(arg sim)" prefix="$(arg prefix)" />

  <joint name="arm_mount_joint" type="fixed">
    <origin xyz="$(optenv HUSKY_KINOVA_XYZ -0.105 0.0 0.0)" rpy="$(optenv HUSKY_KINOVA_RPY 0 0 0)" />
    <parent link="top_plate_front_link" />
    <child link="kinova_arm_base_link" />
  </joint>
</robot>
```

To include this in the Husky model, just set the `HUSKY_URDF_EXTRAS` variable in your setup file (or `$HOME/.bashrc`)
```
export HUSKY_URDF_EXTRAS=/path/to/husky_gen3_description.urdf.xacro`
```
substituting `/path/to` with the actual path to the file.

closed 2024-01-02 by tonybaltovski