husky/husky #292: How to set IP address of wired net work?

Karin-Sugi, 2023-06-05:
Hi, as i mentioned, now I'd like to fix IP address in wired network because the address sometimes changes. I already know the way in wifi network following [this](https://www.clearpathrobotics.com/assets/guides/melodic/husky/HuskyNetwork.html) .

I tried to do that with netplan yaml file, but it doesn't look working...

Any solutions?

Thanks, in advance
Karin

civerachb-cpr, 2023-06-06:
What's likely happening is that your Husky is being connected to a wired DHCP source, and the assigned address is rotating.

The best solution would probably be to look up the Husky's wired MAC address and modify your DHCP server to provide a static address allocation for that MAC.  Most consumer routers have an option to do this somewhere.  If you don't have access to the router's configuration (e.g. because it's a university or company owned router) you may need to go through your IT administrator to request this.  Alternatively, your IT administrator may also be able to assign you a static address outside the normal DHCP range.  In this case, you can modify your netplan file to simply use the fixed address:

```
bridges:
  br0:
    dhcp4: no                   # CHANGE: yes -> no
    dhcp6: no
    interfaces: [bridge_eth, bridge_en]
    addresses:
      - 192.168.131.1/24
      - 10.27.204.101/16        # ADD: this line, with your desired IP address and the netmask size
```

Because DHCP allocations are outside our direct control we cannot provide an exact solution. But hopefully you can take this information to your IT administrator and they can work with you to solve the problem.

Karin-Sugi, 2023-06-15:
Thanks for your reply. The problem is not  the assigned address is rotating, but the assigned address `br0` is never changed when i change network!
Husky is wired to a router module. Recently, I changed robot network router from `192.168.13.xxx` to `192.168.201.xxx`, but the IP address is still first one(192.168.13.xxx) and never changed. I wander why it happened.

Thus, I'd like to change Husky's IP address in wired network. 
Now, I tried to modify netplan file, but I couldn't find any files with `bridges` statement. Couls you tell me the file path?

Thanks in advance,
Karin

civerachb-cpr, 2023-06-19:
If your Husky is old enough that it's using ROS Melodic, it's possible that it was set up using the older `/etc/network/interfaces` file instead of `/etc/netplan/50-clearpath-bridge.yaml`

Please check both of those files to see which (if either) us being used to define `br0`

Karin-Sugi, 2023-06-23:
I found some text about `br0` setting in `/etc/network/interfaces`! I think the `br0` is set automatically at first, but now it never change even after changing the network. 
Do you know the reason? If possible, I'd like to change the `br0` address manually. Is it OK??

The contents of the file is following.
```
auto lo br0 br0:0
iface lo inet loopback

# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex (eth.*)|(en.*)
  address 192.168.13.yyy
  netmask 255.255.255.0
  bridge_maxwait 0

# Also seek out DHCP IP on those ports, for the sake of easily getting online,
# maintenance, ethernet radio support, etc.
iface br0:0 inet dhcp

```

civerachb-cpr, 2023-06-23:
You should _not_ modify `br0`.  All Clearpath robots are configured to use the `192.168.131.0/24` subnet internally to communicate with sensors, sub-controllers, and networking equipment inside the robot.  Some devices will expect the ROS master to be available internally at `http://192.168.131.1:11311`.  Modifying `br0` to change this address could prevent hardware inside your robot from working correctly.

If you are connecting your robot to a wired DHCP source, you can use the `br0:0` interface.  To disconnect & reconnect (or to refresh your DHCP lease) please try connecting the ethernet cable to the Husky's computer and running
```
sudo ifdown br0:0  # this may fail if the interface is already down; don't worry if this happens, it's fine
sudo ifup br0:0
```

If you want to add an additional static interface to the bridge, e.g. you have a cobot with a fixed IP address that the Husky needs to communicate with over ethernet, you can add additional interfaces to the bridge by modifying `/etc/network/interfaces`:

```
# ADD: br0:1 here for our new static interface
auto lo br0 br0:0 br0:1
iface lo inet loopback

# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex (eth.*)|(en.*)
  address 192.168.13.yyy
  netmask 255.255.255.0
  bridge_maxwait 0

# Also seek out DHCP IP on those ports, for the sake of easily getting online,
# maintenance, ethernet radio support, etc.
iface br0:0 inet dhcp

# Configure the new static address here
# Substitute the address and netmask as necessary for your network
iface br0:1 inet static
  address 10.25.0.101
  netmask 255.255.0.0
  bridge_maxwait 0
```

If you are using Ubuntu 18.04 (ROS Melodic) there is a known bug where multiple static addresses on a bridge will not always come up correctly on startup. To work-around this problem, you can use `/etc/rc.local` to force the interfaces up:

`/etc/rc.local` -- create this file if it does not exist
```
#!/bin/bash

# force the secondary static interface up
ifup br0:1

exit 0
```
Make sure the file is executable by running `sudo chmod +x /etc/rc.local` and then reboot your computer.  Running `ifconfig -a` should show both `br0` and `br0:1` as connected with their static addresses (192.168.131.1/24 for `br0` and `10.27.0.101/16` for `br0:1` in the example above).

Karin-Sugi, 2023-06-28:
Thank you for your reply! I could connect wired network with `br0:0` interface.

Now, I have two questions.
Firstly, I am not familiar to bridge network, so I have no idea why I should use another interface `br0:1` to add static address to Husky and which one is the static address (it looks like `10.25.0.101` is, but this is not my network address... ).

Second, I wander why the br0 address is never changed automatically as it did at first time I connected Husky to the `192.168.13.yyy` network (the br0 address is changed into `192.168.13.yyy` automatically). Now I cannot access `192.168.131.1` , instead I can use `192.168.13.yyy` address (ros master uri is also cahnged into this address (perhaps)). 

I really appreciate your support.

civerachb-cpr, 2023-06-28:
The ethernet bridge basically allows all of the physical ethernet ports on the computer to operate concurrently on multiple IP addresses.  When using the older `/etc/network/interfaces` configuration the overall bridge has a name, `br0` in this case, which has the bridge's primary IP address.  Additional addresses can be added as new, numbered interfaces `br0:0` for the first, `br0:1` for the second, `br0:2` for the third, and so on. This is just a convenient way of indicating how many IP addresses the bridge is using, and provides a convenient way to bring indivudual addresses up or down.

In Clearpath robots `br0` is set up at our production facility to use `192.168.131.1` with  subnet mask of `255.255.255.0`. We _strongly_ recommend that this should never be changed, as this address sometimes gets used as the destination IP address for sensor data (e.g. the VLP-16 3D lidar is often configured to send its packets to this address, rather than using a broadcast address which can flood the robot's internal network).

In the `/etc/network/interfaces` file you posted earlier, it looks like the address for `br0` _was_ changed:
```
# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex (eth.*)|(en.*)
  address 192.168.13.yyy
  netmask 255.255.255.0
  bridge_maxwait 0
```
I don't know why it was changed, nor who changed it. But I would recommend changing the address back to `192.168.131.1`. If you need an address on the `192.168.13.0/24` subnet, add an additional interface to the bridge (e.g. `br0:1` like I showed in the previous post.

The `br0` address never changes because it is configured to be `static`.  This means the IP address is fixed and will never change.

The default `br0:0` configuration uses DHCP, so that address may change depending on what address is assigned to the Husky from your DHCP server.

Because this issue is not actually related to the Husky's ROS packages I'm going to close it. I recommend you contact our support team (I posted the link in another ticket you created asking about the wireless network management). Please provide a link to this thread when you contact our support team so they'll have all the relevant background information.