husky/husky #293: Cannot share ros master!

Karin-Sugi, 2023-06-12:
Hi, I'm trying to share ros master between Ubuntu on Husky <==> RaspberrPi or my remote laptop. 
Now, I can see all topics with `rostopic list` command and publish some messages to the Husky, but cannot subscribe topic sent from Husky PC with `rostopic echo` command even when master share between raspberrypi <==> my remote pc works fine...

How can I solve it?

Thanks in advance.
Karin

Karin-Sugi, 2023-06-13:
Thanks, but I don't use hostname. I ran following command at every computer.
```
$ export ROS_MASTER_URI=http://192.168.xxx.yyy:11311 (Husky's ros master)
$ export ROS_IP=192.168.xxx.zzz (each PC's IP address)
```

I think this is due to the Husky's service setting which starts `sudo systemctl start ros`. 

My husky is set to start this servese automatically after turning on, and the launch file I launched after input of ROS_MASTER_URI worked fine. I mean only topics launched automatically (like `husky_velocity_controller/cmd_vel` topic) don't work (other PC cannot subscribe those topics).

Do you have any setting of ROS_MASTER_URI in your service setting? Even when I restart the service after setting maser uri with `systemctl stop ros` and `systemctl start ros`, the result doesn't change.

civerachb-cpr, 2023-06-13:
If you open `/usr/sbin/ros-start` you can see the contents of the script that runs when the `ros.service` systemd job executes. There's normally a line in this file that sets the Husky's `ROS_MASTER_URI` to `http://127.0.0.1:11311` and sets the `ROS_IP` and `ROS_HOSTNAME` environment variables.

What _exactly_ is the error you're getting when you try to publish or subscribe to the Husky topics from your remote computer?

Karin-Sugi, 2023-06-15:
The problem my remote computer cannot subscribe Husky's topics even though it can see Husky's topic list and publish some messsage to the Husky's topic.

I know it is quite strange situation. Actuallly, I eager to use husky's imu data. That's why I'd like to fix this problem and be able to subscribe Husky's topics.

This problem occurs in only Husky's topics ( especially Husky's default topics ), so I thought it is because of service setting... As you said, I found `ROS_MASTER_URI` is set in `/usr/sbin/ros-start`. I modified its file, but situation is never changed. Thus what I did and tis result are following.
```
1. change ROS_MASTER_URI
    export ROS_MASTER_URI="http://192.168.201.xxx:11311"     (Husky's wifi IP address: `wlan0`)

2. echo ROS_MASTER_URI
    http://192.168.201.yyy:11311    (Husky's wired IP address: `br0`)

3. plug out LAN cable
    http://192.168.201.yyy:11311
    
```
Why such a strange situation is occuring? and how can i fix it?

civerachb-cpr, 2023-06-19:
We do not recommend changing the ROS_MASTER_URI inside `/usr/sbin/ros-start`; the Husky should use a `roscore` process running locally inside the robot. Using an external master may result in performance problems.

On your laptop you should be setting the ROS_MASTER_URI envar to point to the Husky's hostname or external IP address.  The laptop and robot should be on the same LAN.

Please review the link to the Jackal documentation I provided earlier.  All the instructions for setting up a remote ROS connection are there.

You also need to make sure you have the necessary messages installed & sourced on your laptop.  For example, to read the IMU data you will need the `sensor_msgs` package installed on your laptop.  At minimum you should have these packages installed:

- std_msgs
- sensor_msgs
- geometry_msgs
- husky_msgs

These can be installed by running
```
sudo apt-get install ros-noetic-std-msgs ros-noetic-sensor-msgs ros-noetic-geometry-msgs ros-noetic-husky-msgs
```

Note that to install `husky_msgs` you may also need to follow the instructions at http://packages.clearpathrobotics.com to add Clearpath's package servers.

Karin-Sugi, 2023-06-23:
I tried again, but result is the same.

> On your laptop you should be setting the ROS_MASTER_URI envar to point to the Husky's hostname or external IP address. The laptop and robot should be on the same LAN.
Yes. I set `ROS_MASTER_URI` to point to the Husky's external IP address, where I can do ssh from my laptop, and of cource my laptop and robot are in the same network (I can do ssh each other).

However, when I tried `rostopic list`, following error message was returned.
```
Value error: invalid master URI: 192.168.201.xxx
The traceback for the exception was written to the log file
```

As I said, I think this is because original husky ros packages are run in other master.
The reason why I assume so is when I type `echo $ROS_MASTER_URI` soon after turning on Husky, the MASTER_URI is set `br0` address.

civerachb-cpr, 2023-06-23:
It looks like you aren't setting the port number and protocol for your ROS master URI. The `ROS_MASTER_URI` envar should be `http://192.168.201.xxx:11311`.  From the error it looks like you just set it to the IP address, omitting the `http://` at the start and the `:11311` at the end.

In the `ros-start` script the master is set to the loopback IP address (`127.0.0.1`, aka `localhost`) because the master is running locally on the Husky.  This is normal, and should not be modified.

When connecting your laptop to the Husky's ROS master you should run
```
export ROS_MASTER_URI=http://192.168.201.xxx:11311
```
replacing `.xxx` with the final stanza of the robot's IP address. You _must_ include `http://` at the start and `:11311` at the end.

Karin-Sugi, 2023-06-28:
Oops, you are right.

Now, my situation is that I can send topic (like `/husky_velocity_controller/cmd_vel`) from my remote pc to Husky, but the remote cannot subscribe the topic (where I could send messages) from Husky. Do you have any idea?

This was happened even when I connect my remote PC to Husky **directly** (not only through wifi) with echoing `ROS_MASTER_URI` like below on each terminal.
```
export ROS_MASTER_URI=http://192.168.13.7:11311
``` 
The address `192.168.13.7` is `br0` address of my Husky, and when I check the `ROS_MASTER_URI` soon after Husky is launched, the same URI is returned.

civerachb-cpr, 2023-06-28:
From what you posted in your other ticket, it looks like you've replaced the `br0` address with 192.168.1.7.  The `br0` network is normally contained _within_ the robot, and isn't accessible from outside unless you've connected an ethernet cable directly between your laptop and the Husky.

If you are communicating with the Husky over the wireless, you should be using the Husky's wireless IP address as the ROS_MASTER_URI, not the wired address from `br0`.

Are you able to ping the Husky from your laptop?
```
ping 192.168.13.7
```
and can you SSH into the robot too:
```
ssh administrator@192.168.13.7
```

Karin-Sugi, 2023-07-10:
Thanks for your reply!
Fiirst of all, I NEVER replaced `br0` address. It was changed AUTOMATICALLY.

When I tried to connect Husky over wireless network, of course I usually tried to access Husky's wireless address, which doesn't work either. I mean I cannot access Husky's ros master not only over the wireless network but also in the wired (of course, I change master's IP adderess based on the network).

Also, I can access Husky with both ping and ssh over the wireless & wired network. While I connect Husky with ssh to check ros master URI and all topics fine in Husky computer, my remote pc cannot subscribe Husky's topic.

Thanks in advance,
Karin

civerachb-cpr, 2023-07-10:
It sounds like you may either have misconfigured your network, or are not properly setting the ROS_MASTER_URI.  In either case this is not a bug with the Husky ROS nodes themselves, which is what github issues in this repo are for.

Please contact Clearpath's customer support (linked in your previous related issues).  They'll be able to provide further debugging steps to help resolve your networking problems.

closed 2023-07-10 by civerachb-cpr

Karin-Sugi, 2023-07-12:
If I either have misconfigured network, or am not properly setting the ROS_MASTER_URI as you said, why can i publish message to my Husky through the topic?? The published messages made Husky move correctly.