husky/husky #297: Resizing Husky Model in Gazebo

derektan95, 2023-08-09:
Hi, 

**Describe the the feature you would like**
Is it possible to resize the husky model in gazebo, to better match the size of other 4 wheel robots? (i.e. possibly via husky.urdf.xacro)? Naively setting the `scale` param within the `<mesh filename="package://husky_description/meshes/base_link.dae" scale="0.1 0.1 0.1" />` does not work. Thanks!

civerachb-cpr, 2023-08-09:
The size of the Husky model is intended to accurately reflect the physical Husky robot (which only comes in one size).

Can I ask why you want to resize the model?

derektan95, 2023-08-15:
Hi @civerachb-cpr, 

Awesome idea. This will work well. Thanks!

closed 2023-08-15 by derektan95

civerachb-cpr, 2023-08-15:
You're welcome! Glad to hear the Jackal will suit your needs.