husky/husky #304: ROS2 diagnostic publisher

cedricpradalier, 2024-01-13:
Hello,

We are porting our husky to ROS2 and I just tested the humble-devel branch. This seems to work great for the controller but I could not see where the diagnostic messages defined in husky_base/src/diagnotics.cpp are published. Is there a plan to add a publisher for this diagnostic array ? It seems to be the only place where the internal sensor data (power, voltage, current, ...) are published. 

If needed, I could implement it myself, but I could use some pointers to get started. 

Thanks.

tonybaltovski, 2024-01-13:
Hi @cedricpradalier, for ROS 2 Humble, the Husky code base is here: https://github.com/clearpathrobotics/clearpath_common.

cedricpradalier, 2024-01-13:
Thanks. Will it be compatible with the old horizon firmware we have on our
husky?

tonybaltovski, 2024-01-15:
The communication protocol is the same for all.