husky/husky #310: How to add sensors to sensor arch ？

liugehaizaixue, 2024-06-01:
ubuntu 20.04
ros: noetic
branch : noetic-devel

I installed the instructions to start LASER-3D, SENSOR-ARCH, and REALSENSE.
But why do LASER-3D and REALSENSE not appear normally on SENSOR-ARCH, but on the Plate.
Actually, I have tested many versions of the code and none of them are displaying properly?
Did I miss any steps?

```
export HUSKY_LASER_3D_ENABLED=1
export HUSKY_SENSOR_ARCH=1
export HUSKY_REALSENSE_ENABLED=1

roslaunch husky_gazebo empty_world.launch
``` 

[image]

civerachb-cpr, 2024-06-03:
If you add
```
export HUSKY_REALSENSE_PARENT=sensor_arch_mount_link
```
to your setup that will change the RealSense's mount to be the top centre of the arch:
[image]

If you need to offset/rotate the camera you can use
```
export HUSKY_REALSENSE_XYZ="0.0 0.0 0.0"
export HUSKY_REALSENSE_RPY="0.0 0.0 0.0"
```
changing the "0.0" for the necessary X/Y/Z offset (in meters) and Roll/Pitch/Yaw rotations (in radians).

liugehaizaixue, 2024-06-06:
> ```
> export HUSKY_REALSENSE_PARENT=sensor_arch_mount_link
> ```
感谢您的回复！在您的帮助下我解决了问题。

closed 2024-06-06 by liugehaizaixue